earlephilhower/arduino-pico #1446: SD card library doesn't support "Logical" partitions, only "Primary" partitions will mount

XZCE, 2023-05-15:
Having spent the last 2 days failing to detect any of my SD cards with my "Arduino Pico" sketch (turns out the SPI0 & SPI1 / SD Init was working, but the filesystem mount was not, but without debug turned on they looked like the same issue of failing hardware) I did the following to a couple of my SD cards: (where disk 1 is my SD card - use "list disk" to see what yours are)

diskpart
select disk 1
clean
Open "Computer Management" and create a "New Simple Volume..." and follow the usual process to format.

They now work with my sketch.

Looking at my other non-working cards with Partition Wizard, they are set up as "Logical" partitions instead of "Primary" ones like my newly formatted and now working SD cards.  Hence my post here :)

To save people's time, I would recommend this issue stay open until the library here supports all types of SD card partitions, but I'll understand if this issue is closed - as I looked at the library code (and added lots of my own debugging output statements) but there's little chance I would ever be able to code up a c++ solution to this - I'm C only unless it's simple class stuff - so I don't expect anyone else to work on this.

lyusupov, 2023-05-15:
Most of Arduino Cores (including this one) are using a genuine or a derivative of this library: https://github.com/greiman/SdFat
It would make more sense to file your issue there.

earlephilhower, 2023-05-15:
Thanks, @lyusupov, agreed.  I just took all grieman's work and did the needed ports/edits/etc. to make it work here.  For new features, upstream is where this bug belongs.

closed 2023-05-15 by earlephilhower

XZCE, 2023-05-16:
Sleep well tonight gentlemen...

Even being closed, this issue's details will no doubt help another.  I know, because I searched all the issues here before posting - and I saw at least 1 person just give up.  Though the argument given by the owner in the link to "Greiman" is more along the lines of "I can't be bothered" rather than the SD card optimisation will be compromised.  Really?  Anyway, at least the mention here links things nicely - thanks Supe.

But, incidentally, that link has very little to do with my single partition FAT32 non-GPT SD card(s) which I didn't "clean" first and by-the-way "just" came to me this way.

Also, I would have used any "ported" error code variable access before doing my debug statements, but while they are buried deep down there, I didn't see them being made available to use...  "I just took .. and did the needed ports" == same as that other guy who's finished now.