earlephilhower/arduino-pico #1455: undefined reference to '__holdUpPenSV' in FreeRTOS/port.c

doctea, 2023-05-19:
Hi, linker complains about "FreeRTOS/port.c.o: in function `isr_systick':
port.c:(.time_critical.isr_systick+0x30): undefined reference to `__holdUpPendSV'"

Saw in commit log that this shouldn't be necessary anymore, removed it in my local copy and it links now.

earlephilhower, 2023-05-19:
You needed a `git submodule update`, I think.  Just `git pull`ing would not actually update to the changed FreeRTOS external submodule.

Deleting and restarting from scratch would give the same effect, as you saw.

closed 2023-05-19 by earlephilhower

doctea, 2023-05-20:
Thanks @earlephilhower!  Actually, I did not mean that I solved it by deleting and reinstalling -- I just commented out the remaining use of the variable.  Deleting the framework-arduinopico directory and allowing vscode/platformio to reinstall it actually just fails for me with 'There is not enough space on the disk', even though I have gigabytes free, but I guess that is a separate problem..

earlephilhower, 2023-05-20:
> 'There is not enough space on the disk', even though I have gigabytes free, but I guess that is a separate problem..

See https://arduino-pico.readthedocs.io/en/latest/platformio.html#important-steps-for-windows-users-before-installing if you're under Windows. The Pico-SDK TinyUSB has some paths that break 255 bytes which cause strange errors like this.

doctea, 2023-05-20:
> See https://arduino-pico.readthedocs.io/en/latest/platformio.html#important-steps-for-windows-users-before-installing if you're under Windows. The Pico-SDK TinyUSB has some paths that break 255 [...]

I'd already run into that issue of 'Filename too long', and taken those steps, which solved my encounter with that issue.  Maybe I've missed a setting however.  But updating submodules from the cli (wsl in this case, but on the /mnt/ ntfs mountpoint) seemed to work OK, and I'm happily compiling again :)