earlephilhower/arduino-pico #1460: Error compiling ethernet example for w5500-evb-pico RP2040 on win10

sdrshnptl, 2023-05-20:
I'm trying to compile Ethernet example for w5500-evb-pico on win10 with following platformio.ini
Any idea why this happening?

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower

lib_deps = 
    https://github.com/sdrshnptl/EthernetWebServer.git
```

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.9.0+sha.0c33219) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES:
 - framework-arduinopico @ 1.30101.0+sha.e95248a
 - tool-rp2040tools @ 1.0.2
 - toolchain-rp2040-earlephilhower @ 5.100300.230216 (10.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 51 compatible libraries
Scanning dependencies...
Dependency Graph
|-- EthernetWebServer @ 2.4.1+sha.dcf4d57
|-- ArduinoHttpClient @ 0.4.0+sha.6dc4867
|-- Ethernet_Generic @ 2.8.1
|-- SPI @ 1.0
Building in release mode
Compiling .pio\build\pico\src\main.cpp.o
Compiling .pio\build\pico\lib27f\HTTPClient\HTTPClient.cpp.o
Compiling .pio\build\pico\lib27f\HTTPClient\base64.cpp.o
Compiling .pio\build\pico\liba86\EthernetWebServer\libb64\cencode.c.o
Compiling .pio\build\pico\FrameworkArduino\SerialPIO.cpp.o
In file included from src\main.cpp:3:
src\defines.h:100:2: warning: #warning Use RPI-Pico RP2040 architecture with custom SPI or SPI1 [-Wcpp]
  100 | #warning Use RPI-Pico RP2040 architecture with custom SPI or SPI1
      |  ^~~~~~~
In file included from C:\Users\sdrshnptl\.platformio\packages\framework-arduinopico\libraries\HTTPClient\src\HTTPClient.cpp:28:
C:\Users\sdrshnptl\.platformio\packages\framework-arduinopico\libraries\HTTPClient\src\HTTPClient.h:32:10: fatal error: WiFiClient.h: No such file or directory

********************************************************************
* Looking for WiFiClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClient.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClient.h
*
********************************************************************

   32 | #include <WiFiClient.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\lib27f\HTTPClient\HTTPClient.cpp.o] Error 1
C:\Users\sdrshnptl\.platformio\packages\framework-arduinopico\libraries\HTTPClient\src\base64.cpp: In static member function 'static arduino::String base64::encode(const uint8_t*, size_t, 
bool)':
C:\Users\sdrshnptl\.platformio\packages\framework-arduinopico\libraries\HTTPClient\src\base64.cpp:42:23: error: 'base64_encode_expected_len_nonewlines' was not declared in this scope; did 
you mean 'base64_encode_expected_len'?
   42 |                     : base64_encode_expected_len_nonewlines(length)) + 1);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       base64_encode_expected_len
C:\Users\sdrshnptl\.platformio\packages\framework-arduinopico\libraries\HTTPClient\src\base64.cpp:50:13: error: 'base64_init_encodestate_nonewlines' was not declared in this scope; did you mean 'base64_init_encodestate'?
   50 |             base64_init_encodestate_nonewlines(&_state);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             base64_init_encodestate
*** [.pio\build\pico\lib27f\HTTPClient\base64.cpp.o] Error 1
=============================================================================== [FAILED] Took 17.55 seconds ===============================================================================
```

earlephilhower, 2023-05-20:
Sorry, this is a library issue and not related to the core.  You're using a 3rd party Ethernet, HTTPClient, and WebServer.   Not the core's versions of them AFAICT from your logs.  I suggest paring things down to a single 3rd party lib at a time, and if those fail opening bugs in the appropriate repo.  Good luck!

closed 2023-05-20 by earlephilhower

maxgerhardt, 2023-05-21:
PIO + regular Ethernet library is definitely possible, we discussed that in https://github.com/earlephilhower/arduino-pico/discussions/1169 with e.g. the sketch https://gist.github.com/maxgerhardt/a33206a8e76ee6c0b66d574c4a1ad1e8.

So https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebServer/WebServer.ino should work too.

maxgerhardt, 2023-05-21:
>C:\Users\sdrshnptl\.platformio\packages\framework-arduinopico\libraries\HTTPClient\src\base64.cpp:42:23: error: 'base64_encode_expected_len_nonewlines' was not declared in this scope; did 
you mean 'base64_encode_expected_len'?
   42 |                     : base64_encode_expected_len_nonewlines(length)) + 1);

Just a heads up on why that error is happening: The EthernetWebServer also has `src/libb64/base64.h` which interferes with the base64.h found in this core, so that's a game-over. 

I've corrected the library and uploaded a minimally compiling project.

https://github.com/maxgerhardt/pio-wiznet_5500_evb_pico-webserver

I also corrected the `platformio.ini` to `board = wiznet_5500_evb_pico`, because why use `pico` when we have dedicated board definitions for it?

Does it work during runtime? I don't know, I don't have that board.

sdrshnptl, 2023-05-21:
> I also corrected the `platformio.ini` to `board = wiznet_5500_evb_pico`, because why use `pico` when we have dedicated board definitions for it?

i should have used this correctly.
Thanks @maxgerhardt !