earlephilhower/arduino-pico #1463: RP2040 combined HID libraries keyboard +mouse + absolute mouse not working

silpakcx, 2023-05-22:
Hi... Thanks for enabling absolute mouse support in 3.1.1 release. 
I have to use keyboard, mouse and absolute mouse libraries for my project (because Absolute mouse library only supports mouse move, no other mouse actions). But when I used these three libraries together absolute mouse is not working as expected. I have seen the issue ( https://github.com/earlephilhower/arduino-pico/issues/1410 ) and done the fix also. Now keyboard and mouse libraries are working fine but Absolute mouse library is not working (mouse is moving but not to absolute positions). 
I have used keyboard and absolute mouse libraries together which also works fine for me. But the issue comes when i try to use mouse and absolute mouse together.

Can we use mouse and absolute mouse libraries together? If not, please add other mouse actions like mouse scroll and clicks to absolute mouse library. 

version : 3.1.1
Arduino IDE : 1.8.19
Sample code :

`#include <MouseAbsolute.h>
#include <Keyboard.h>
#include <Mouse.h>

String user_input;

void setup() {
        Serial.begin(9600);
        MouseAbsolute.begin();
        Keyboard.begin(KeyboardLayout_en_US);
        Mouse.begin();

}

void loop() {
        if (Serial.available() > 0) {

                user_input = Serial.readString();
                if (user_input == "INPUT_1\r\n") {
                        MouseAbsolute.move(1720, 1885, 0);
                        delay(10);
                }

                else if (user_input == "INPUT_2\r\n") {
                        MouseAbsolute.move(25334, 18789, 0);
                        delay(10);
                }

                else if (user_input == "INPUT_3\r\n") {
                        Keyboard.press('A');
                        delay(10);
                        Keyboard.releaseAll();
                }
                else if (user_input == "INPUT_4\r\n") {
                        Mouse.press(MOUSE_RIGHT);
                        delay(10);
                        Mouse.release(MOUSE_RIGHT);
                }

        }
}`

earlephilhower, 2023-05-22:
`Mouse` and `MouseAbsolute` share a HID entry, so you can only use one or the other now.  So, what you're seeing is the expected and intended behaviour with only one appearing.

However, `MouseAbsolute` should support `click`/etc.  Same as plain `Mouse`.  It's subclasses off of the HID_Mouse, same as `Mouse`

Did you try calling `MouseAbsoule::click()` and have it fail to compile?   If so, please give an MCVE.

earlephilhower, 2023-05-25:
Thanks for checking back in.  Glad it's working!