earlephilhower/arduino-pico #1464: Reinstalling packages gives 'No space left on device' errors, even though disk is not full

doctea, 2023-05-22:
Hi, I'm seeing this problem on two different computers now; platformio in vscode:

Thought it might be the "long paths" problem, but when I ran into that originally the error message was specific to the paths being too long, and I have enabled long paths in the OS and every every git system environment I can think of (Windows regedit, git cmd, powershell - I'm not sure where platformio takes its git environment from for the installer tasks though?) and still get these errors.  

Also, the filepaths that it seems to fail on are nowhere near the 260 character limit (more like 140 ish), so I feel like that might be a red herring here.

I worked around this on the first computer I encountered it on by git pulling + updating submodules from cli, but this second system is complaining about local files that would be overwritten if I attempt that.

Portions of the output log below:-

```Resolving pcb dependencies...
Tool Manager: Installing git+https://github.com/earlephilhower/arduino-pico.git#e95248a78723f6bfebb3d0ab0834a0dd216ad81a
git version 2.40.1.windows.1
Cloning into 'C:\Users\Rad\.platformio\.cache\tmp\pkg-installing-pq5q0s0y'...

Updating files:  24% (210/858)
Updating files:  25% (215/858)
Updating files:  26% (224/858)

[...etc...many 'Submodule registered for path', etc... ]

Submodule path 'tools/pyserial': checked out '0e7634747568547b8a7f9fd0c48ed74f16af4b23'
Error: Traceback (most recent call last):
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\cli.py", line 70, in invoke
    return super().invoke(ctx)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\project\commands\init.py", line 104, in project_init_cmd
    install_project_dependencies(
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 106, in install_project_dependencies
    already_up_to_date = not install_project_env_dependencies(env, options)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 131, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 148, in _install_project_env_platform
    PlatformPackageManager().install(
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 62, in install
    p.install_required_packages(force=force)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 76, in install_required_packages
    self.install_package(name, force=force)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 70, in install_package
    return self.pm.install(spec or self.get_package_spec(name), force=force)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 47, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_uri(spec.uri, spec)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 200, in install_from_uri
    return self._install_tmp_pkg(pkg_item)
  File "C:\Users\Rad\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 292, in _install_tmp_pkg
    shutil.copytree(tmp_pkg.path, dst_pkg.path, symlinks=True)
  File "C:\Users\Rad\.platformio\python3\lib\shutil.py", line 566, in copytree
    return _copytree(entries=entries, src=src, dst=dst, symlinks=symlinks,
  File "C:\Users\Rad\.platformio\python3\lib\shutil.py", line 522, in _copytree
    raise Error(errors)
shutil.Error: [('C:\\Users\\Rad\\.platformio\\.cache\\tmp\\pkg-installing-pq5q0s0y\\pico-sdk\\lib\\tinyusb\\hw\\mcu\\nxp\\mcux-sdk\\devices\\MIMX8DX6\\MIMX8DX6_cm4.h', 'C:\\Users\\Rad\\.platformio\\packages\\framework-arduinopico\\pico-sdk\\lib\\tinyusb\\hw\\mcu\\nxp\\mcux-sdk\\devices\\MIMX8DX6\\MIMX8DX6_cm4.h', '[Errno 28] No space left on device'), ('C:\\Users\\Rad\\.platformio\\.cache\\tmp\\pkg-installing-pq5q0s0y\\pico-sdk\\lib\\tinyusb\\hw\\mcu\\nxp\\mcux-sdk\\devices\\MIMX8DX6\\scfw_api\\svc\\pad\\pad_rpc_clnt.c', 'C:\\Users\\Rad\\.platformio\\packages\\framework-arduinopico\\pico-sdk\\lib\\tinyusb\\hw\\mcu\\nxp\\mcux-sdk\\devices\\MIMX8DX6\\scfw_api\\svc\\pad\\pad_rpc_clnt.c', '[Errno 28] No space left on device'), ('C:\\Users\\Rad\\.platformio\\.cache\\tmp\\pkg-installing-pq5q0s0y\\pico-sdk\\lib\\tinyusb\\hw\\mcu\\nxp\\mcux-sdk\\devices\\MIMX8DX6\\scfw_api\\svc\\pm\\pm_api.h', 'C:\\Users\\Rad\\.platformio\\packages\\framework-arduinopico\\pico-sdk\\lib\\tinyusb\\hw\\mcu\\nxp\\mcux-sdk\\devices\\MIMX8DX6\\scfw_api\\svc\\pm\\pm_api.h', '[Errno 28] No space left on device'), ('C:\\Users\\Rad\\.platformio\\.cache\\tmp\\pkg-installing-pq5q0s0y\\pico-sdk\\lib\\tinyusb\\hw\\mcu\\nxp\\mcux-sdk\\devices\\MIMX8DX6\\scfw_api\\svc\\pm\\pm_rpc.h', 'C:\\Users\\Rad\\.
[...]
etc
```

earlephilhower, 2023-05-22:
Just to triple check, did you complete everything in https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst#important-steps-for-windows-users-before-installing (including the reboot after changing the registry)?

doctea, 2023-05-22:
> Just to triple check, did you complete everything in https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst#important-steps-for-windows-users-before-installing (including the reboot after changing the registry)?

Hmm, for some reason gpedit.msc wasn't available on my other computer; but when I opened up regedit to `HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem`, `LongPathsEnabled` was already set to 1.  I've just checked in gpedit on this computer and it was disabled, so I've enabled it there.

I'll try a reboot on t'other computer tomorrow, and when next convenient I'll attempt a reboot+reinstall of the package on this computer, too.

My apologies if this turns out to be another wild goosechase!

maxgerhardt, 2023-05-23:
`framework-pico` on my system needs over 3.78GB **on disk** space.

[image]

Most of it in the `.git` folder due to tinyusb, and `pico-sdk` due to tiny-usb. It's really have since this pico-sdk versions uses a TinyUSB version that has all SDKs for MCUs submodules and PlatformIO pulls them all. The next version will have that fixed already. But for now, you'll need more diskspace.

earlephilhower, 2023-05-23:
Seems like it really is an out of space error here.  Closing.   Hopefully Pico-SDK 1.5.1 or 1.6 will pull in the updated TinyUSB.

closed 2023-05-23 by earlephilhower

doctea, 2023-05-23:
wow, who'd've thought such a small computer would need such large development files!  thanks for figuring this out for me.  

both my systems were previously even lower on disk space than they are now when they were originally installed, which is why these errors now have caught me so by surprise.  sounds like there has been an increase in space requirements due to TinyUSB too, which may account for why i never ran into this problem when i first installed things a few months back.

thanks again!