earlephilhower/arduino-pico #1465: Incorrect I2C port assignments on Adafruit RP2040 Feather

emeb, 2023-05-23:
The Adafruit RP2040 Feather board assigns both the SCL/SDA pins on its header and Qwiic connector to RP2040 pins 2,3 which is defined as Wire1 in the file `variants/adafruit_feather/pins_arduino`.
```
// Wire
#define PIN_WIRE0_SDA  (24u)
#define PIN_WIRE0_SCL  (25u)
#define PIN_WIRE1_SDA  (2u)
#define PIN_WIRE1_SCL  (3u)
```
As a result it's necessary to override the default port value in most all interfacing examples which assume Wire/Wire0. This is especially problematic for the miniTFTwing which does not allow overriding the default port in its constructor.

emeb, 2023-05-23:
This issue has also been reported to the Adafruit Seesaw library: https://github.com/adafruit/Adafruit_Seesaw/issues/83

anecdata, 2023-05-23:
The Feather [Specification](https://learn.adafruit.com/adafruit-feather/feather-specification) indicates:
> SDA & SCL - this should be your main I2C bus.
(referring to the pins with the "SDA" and SCL" silkscreen, the two at the bottom of the short side of the feather headers)

Those pins on other Feather MCUs are `Wire` rather than `Wire1`.

I think the confusion comes in with the RP2040 terminology `SCL0` and `SDA0` vs. `SCL1` and `SDA1` designations on potential GPIO pins that can use one or the other of the internal DW I2C peripherals. Many GPIO pins can potentially be used as the 0 peripheral or the 1 peripheral, and this is a separate concept from and default or secondary I2C bus.

earlephilhower, 2023-05-23:
Can you explain what pins should be mapped to `Wire` and which should go to `Wire1`?  It might be a matter of simply using the swapping capability https://github.com/earlephilhower/arduino-pico/blob/3b4fb295ea0f7ef4b45ede63b12352b7a42bd82c/libraries/Wire/src/Wire.cpp#L401-L409

emeb, 2023-05-23:
These are the pins that are actually connected to I2C functions on the RP2040 Feather:
```
#define PIN_WIRE1_SDA  (2u)
#define PIN_WIRE1_SCL  (3u)
```
Pins 24 and 25 which are the current defaults (WIRE0) are actually used for SWD programming and only connect to some testpoints and a programming header.

emeb, 2023-05-23:
I think so. It seems that you need to change both the WIREX_DEVICE and and PIN_WIRE_ definitions, but that ought to do it.

caternuson, 2023-05-23:
The Feather RP2040 variant predates the current BSP's ability to allow I2C pin mapping, which was added here:
https://github.com/earlephilhower/arduino-pico/pull/606
So the use of `I2C1` ended up being `Wire1`.

This could be changed, but it would be breaking. All current code using `Wire1` would then break.

earlephilhower, 2023-05-23:
I think this probably should be fixed, but needs a big notice in release notes.  Other Adafruit boards are using the redefined Wire/Wire1 and this is an outlier just because it came early...

closed 2023-05-23 by earlephilhower

jfentonatnklabs, 2023-07-26:
This seems to break the peripheral interrupts(onReceive callback function never gets called), causing the RX buffer to fill, and SCL to be pulled low permanently causing the controller to freeze. TalkingToMyselfExample.ino does not work (even after swapping the Wire<>Wire1) assignments. Reverting back to the old pin_arduino.h fixes it. Not sure what the dependency is.

earlephilhower, 2023-07-26:
@jfentonatnklabs can you please give #1607 a try and report back if it fixes things for you?  IT seems straightforward, but I obviously don't have a Feather board to test on.

earlephilhower, 2023-07-26:
Err, that's #1608, not #1607...

PR #1468 (2023-05-23): BREAKING: Swap Wire1 and Wire in Adafruit Feather
Fixes #1465
commit cac9eb0c (2023-05-23): BREAKING: Swap Wire1 and Wire in Adafruit Feather (#1468)

Fixes #1465

The Adafruit Feather came onboard before the Wire swapping was supported
in the core, so it's backwards from the real definition.  Now that swapping is
supported, fix it to match the rest of the boards.