earlephilhower/arduino-pico #1475: Priority 7 FreeRTOS task hangs on Pico W when writing to LED_BUILTIN LED

dlkeng, 2023-05-25:
In the process of porting the FreeRTOS Comprehensive Demo to Arduino-Pico on a Raspberry Pi Pico W, I uncovered an issue with writing to the LED_BUILTIN LED on the Pico W (see: https://freertos.org/smp-demos-for-the-raspberry-pi-pico-board.html). This problem appears to hang the priority 7 FreeRTOS task that is controlling the LED while another lower priority FreeRTOS task is doing semaphore operations. This problem does not occur if there are no semaphore operations occurring in another task. This problem does not occur on the non-W Pico.

This is compiled on a Windows 10 Pro PC with version 1.8.13 of the Arduino IDE and using V3.2.0 of Arduino-Pico. The board settings are shown in the attached image below.

Two observations have been made about this issue:

1. Running the task at a lower priority does not have this problem
2. Controlling an LED on any other standard GPIO pin does not have this problem

The following is a shortened Arduino sketch to demo the problem:
```
/*
    Demonstrates Pico W FreeRTOS priority 7 bug with LED_BUILTIN LED.

Defective Serial Output:
    FreeRTOS_PicoW_Bug.ino
     Starting test:

    Toggle LED
    LED On

The FreeRTOS priority 7 "Toggle" task hangs writing to the LED_BUILTIN LED.

ID, NAME,   STATE,  PRIO, STACK
 0: Toggle  Ready    7    246
 1: CORE0   Running  4    902
 2: IDLE1   Ready    0    230
 3: Test    Ready    0    216
 4: IDLE0   Ready    0    229
 5: USB     Blocked  7    186
 6: Tmr Svc Blocked  2    989

----------------------------------------------------------------------------------
Expected Correct Serial Output:
    FreeRTOS_PicoW_Bug.ino
     Starting test:

    Toggle LED
    LED On
    Toggle LED Done!

    Toggle LED
    LED Off
    Toggle LED Done!

    Toggle LED
    LED On
    Toggle LED Done!

    Toggle LED
    LED Off
    Toggle LED Done!

    ...
    ...
    ...

The FreeRTOS priority 6 "Toggle" task continues writing to the LED_BUILTIN LED.

ID, NAME,   STATE,  PRIO, STACK
 0: Toggle  Ready    6    246
 1: CORE0   Running  4    902
 2: IDLE1   Ready    0    230
 3: Test    Ready    0    216
 4: IDLE0   Ready    0    229
 5: USB     Blocked  7    186
 6: Tmr Svc Blocked  2    989
*/

#include <FreeRTOS.h>
#include <task.h>
#include "semphr.h"

#define DEBUG_LED_PIN           15
#define STATUS_LED_PIN          3

#define LED_TASK_PERIOD         pdMS_TO_TICKS(3000UL)

#define TEST_TASK_PRIORITY      (configMAX_PRIORITIES - 1)      // priority 7
//#define TEST_TASK_PRIORITY      (configMAX_PRIORITIES - 2)      // priority 6

#define TEST_LED                LED_BUILTIN   // at priority 7, this can hang on Pico W !!!???
//#define TEST_LED                STATUS_LED_PIN  // at priority 7, this works on Pico W

SemaphoreHandle_t xSemaphore;

void setup()
{
    Serial.begin(115200);
    delay(2000);                 // allow time for Arduino's serial window to re-connect
    Serial.println();
    Serial.println("FreeRTOS_PicoW_Bug.ino");

    pinMode(TEST_LED, OUTPUT);
    digitalWrite(TEST_LED, LOW);

    pinMode(DEBUG_LED_PIN, OUTPUT);
    digitalWrite(DEBUG_LED_PIN, LOW);

    Serial.println(" Starting test:");
    xSemaphore = xSemaphoreCreateCounting(1, 0);
    xTaskCreate(CountingSemaphoreTask, "Test", configMINIMAL_STACK_SIZE, NULL, tskIDLE_PRIORITY, NULL);
    xTaskCreate(ToggleLedTask, "Toggle", configMINIMAL_STACK_SIZE, NULL, TEST_TASK_PRIORITY, NULL);
}

void loop()
{
    delay(850);
    digitalWrite(DEBUG_LED_PIN, HIGH);
    delay(150);
    digitalWrite(DEBUG_LED_PIN, LOW);
}

void ToggleLedTask(void * pvParameters)
{
    (void)pvParameters;
    bool led_state = false;

    while (1)
    {
        vTaskDelay(LED_TASK_PERIOD);

        // Toggle the test LED
        Serial.println("");
        Serial.println("Toggle LED");
        if (led_state)
        {
            Serial.println("LED Off");
            digitalWrite(TEST_LED, LOW);
            led_state = false;
        }
        else
        {
            Serial.println("LED On");
            digitalWrite(TEST_LED, HIGH);
            led_state = true;
        }
        Serial.println("Toggle LED Done!");
    }
}

void CountingSemaphoreTask(void * pvParameters)
{
    (void)pvParameters;

    while (1)
    {
        xSemaphoreGive(xSemaphore);
        xSemaphoreTake(xSemaphore, 0);      // 0 = "don't block"
    }
}

```
Arduino IDE Pico W board settings:

[image]

earlephilhower, 2023-05-25:
Sorry, but this is a known limitation.  No CYW43 chip access is supported in tasks (and no WiFi proper is allowed).  The LED on the PicoW is connected to the CYW43, not the Pico.  Only `setup()` and `loop()` should touch the LED.  (On the normal Pico, the LED is a regular GPIO pin and no such restriction exists).

You *might* get away by pinning the LED task to core 0 using the FreeRTOS SMP APIs.  But that's really the limit of things now.

hreintke, 2023-05-28:
@earlephilhower Does this mean that no WiFi/LWIP is possible when freeRTOS is in use ?

earlephilhower, 2023-05-28:
Currently, you are correct @hreintke .  The internal plumbing for the CYW43 needs to be completedly different in FreeRTOS vs bare metal.  The interrupt and async handlers have to be swapped out.

earlephilhower, 2023-06-04:
Closing this as it's behaving "as expected".  A (large, probably painful) enhancement to support CYW43 access while in FreeRTOS is needed to change this (but would also allow WiFi/BT in FreeRTOS).

closed 2023-06-04 by earlephilhower