earlephilhower/arduino-pico #1480: WiFi.mode(WIFI_OFF) turns Bluetooth LE off at the same time

lyusupov, 2023-05-27:
### Release

3.2.1

### To reproduce

Use [iBeacon.ino](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/BTstackLib/examples/iBeacon/iBeacon.ino) sketch with a minor mod:

```
void setup(void) {
  Serial.begin(9600);

WiFi.mode(WIFI_AP);
WiFi.softAP("TestAP", "12345678");

  BTstack.setup();
  BTstack.iBeaconConfigure(&uuid, 4711, 2);
  BTstack.startAdvertising();

WiFi.mode(WIFI_OFF);
}

```

WiFi.mode(WIFI_OFF) makes the sketch not sending the iBeacons anymore.

### Expected behaviour

BLE should work after Wi-Fi is turned off.

earlephilhower, 2023-06-01:
Looking at the Pico SDK, I see there is a way to disable [Bluetooth](https://github.com/raspberrypi/pico-sdk/blob/f396d05f8252d4670d4ea05c8b7ac938ef0cd381/src/rp2_common/pico_cyw43_driver/include/pico/btstack_cyw43.h#L39), but there is no corresponding "disable wifi" call, so I'm not sure this can be fixed here.  Given that the BT core has to be accessed over the main core's SPI bus, it may not be physically possible.

What is your use case for this?

lyusupov, 2023-06-01:
Use case is well described there: https://github.com/lyusupov/SoftRF/wiki/SkyView-Pico

* the system enables both BT and Wi-Fi at boot time, and
* primary use purpose of Wi-Fi  is  for WebUI _(device status and configuration)_, and
* primary use of BT is for **continuous** NMEA _(or GDL90)_ data transfer, and
* if there is no activity over Wi-Fi for 10 minutes - the system turns Wi-Fi off to save approx. 30 mA of battery current

See this section for data of active current measurements: https://github.com/lyusupov/SoftRF/wiki/SkyView-Pico#battery-life

earlephilhower, 2023-06-01:
Thanks, that's an interesting setup.  I can see the obvious benefit of dropping WiFi when not needed.

But, how did you/they measure the "Active Wi-Fi OFF BLE ON" row of the table if there's no way to disable WiFi while BT is running in the core today?  If you were using MicroPython or something, I might be able to look at your code and see how they implemented it in their own C.

lyusupov, 2023-06-02:
> how did you/they measure the "Active Wi-Fi OFF BLE ON"

this measurement was obtained when the Wi-Fi setup sequence were inactivated from boot process

> If you were using MicroPython

SkyView is a multi-platform C/CPP 'sketch':
https://github.com/lyusupov/SoftRF/tree/master/software/firmware/source/SkyView

earlephilhower, 2023-06-05:
As far as I can see, there is no way to turn off the WiFi core and still have the BT core accessible and undisturbed.  If you can find something in the Pico-SDK, please do let me know.  But for now, this seems like it's not possible w/o low-level CYW43 driver coding.

I suggest you try `WiFi.disconnect()` instead of `WiFi.mode(WIFI_OFF)`.  That should give you *about* the same effect while leaving the WiFi core up and running.  No beacons or data should go thru the WiFi power amplifier (probably biggest power user on-chip) when things are disconnected.

closed 2023-06-05 by earlephilhower

lyusupov, 2023-06-05:
According to https://github.com/earlephilhower/arduino-pico/blob/master/libraries/WiFi/src/WiFiClass.cpp  , the sequence is:

WiFi.mode(WIFI_OFF) -> WiFi.end() -> WiFi.disconnect() -> _wifi.end() -> cyw43_deinit(&cyw43_state)

which means that WiFi.mode(WIFI_OFF) has exactly the same effect as WiFi.disconnect().

earlephilhower, 2023-06-05:
Ah, good catch.  Did not see the `wifi.end` when I was eyeballing it just now.  Why not try removing the `wifi.end()` call and testing the power?  If it really does go down, you could throw a PR w/that change since I'm not sure the disconnect really should be completely turning off the WIFI chip anyway...