earlephilhower/arduino-pico #1493: Interrupt and keyboard HID emulation issue

azsde, 2023-06-02:
Hello,

I am having an issue with the Keyboard emulation feature.

What I am trying to achieve is to write a text to a connected device when a push button is pressed:

```
#include <Keyboard.h>

unsigned long buttonTime = 0; // To prevent debounce

void setup() {
  Serial.begin(115200);
  pinMode(11, INPUT_PULLUP);
  attachInterrupt(11, callback, RISING);
  Keyboard.begin();
}

void callback() {
  //Set timer to work for your loop code time   
  if (millis() - buttonTime > 250) {
    //button press ok
    Serial.println("pressed");
    Keyboard.print("hey");
  }
  buttonTime = millis();
}

void loop() {
  Serial.println("Loop");
  // put your main code here, to run repeatedly:
  delay(5000);
}
```

The serial part works, as expected:

```
Loop
Loop
// Button pressed here
pressed
Loop
```

However, this causes the first letter of the string to be printed over and over on my computer, it never stops:

`hhhhhhhh...hhh` 

Am I missing something here ?

Is using Keyboard.print() inside a interrupt routine possible ?

I thank you in advance for your answer.

Regards,

Azsde.

earlephilhower, 2023-06-02:
> Is using Keyboard.print() inside a interrupt routine possible ?

Nope, and you really shouldn't be using `Serial.print()` there either.  IRQs have limited state and having things which can block (like `print()` or USB output is a bad thing and probably not going to work.

What you'll want to do is set a flag in the ISR and return immediately.  Then, your `loop()` can poll the (`volatile`) flag and when it's set, do whatever you need to do.  You can also do things like IRQ-safe queues/etc. for more complicated things.

closed 2023-06-02 by earlephilhower

azsde, 2023-06-02:
Perfect thank you for your quick answer.

Have a great day :-)