earlephilhower/arduino-pico #1494: setting Wire.setSDA and Wire.setSCL wrong causes USB corruption

clacktronics, 2023-06-03:
```
#include "Wire.h"

void setup() {
  Wire.setSDA(5);
  Wire.setSCL(4);
  Wire.begin();

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

I have found a bug that's a bit silly but could be confusing to a beginner (and me!). if you set the SCA, SDA pins wrong it corrupts the USB somehow and turns up as "Unknown USB device:" in Windows 10 . Should it somehow be a compilation error?

Windows 10
Arduino IDE 2.1.0
arduino-pico 3.2.1

earlephilhower, 2023-06-04:
It's not actually corrupting the USB port, it's `panic()`ing the core and stopping all USB operation.  This probably happens in the middle of Windows USB negotiation hence the popup.

There **might** be some way of checking for this at build time, with some very complicated `static_assert` using the `#define`d devices (Wire can be `i2c0` or `i2c1` depending on the board).  It's not something I have the time to hack in, but if it is important to you then I'd be happy to look at any PR you might make for it!

closed 2023-06-04 by earlephilhower

clacktronics, 2023-06-06:
Ok it's not something important, just something I had noticed. If it's difficult it's hardly worth it!