earlephilhower/arduino-pico #1498: pico/stdio_usb.h not found

thirstyice, 2023-06-04:
Trying to do some debugging by printing stdio messages to usb, but the compiler can't find the header.
Weird because it finds pico/stdio_uart.h just fine.

Platformio on Windows 11

earlephilhower, 2023-06-04:
That is not supported, nor needed.  Just use `printf` and set the Debug port via the IDE to `Serial` in order to get `printf` outputs.  You can also use `Serial.printf()` if needed.

closed 2023-06-04 by earlephilhower

thirstyice, 2023-06-04:
I'm looking to get the output of a `panic()` over the usb connection.
Where do I set the debug port in the IDE?

earlephilhower, 2023-06-04:
>  Where do I set the debug port in the IDE?

Err...`Tools->Debug Port`

thirstyice, 2023-06-04:
PlatformIO, not Arduino

thirstyice, 2023-06-05:
Never mind, found it in the documentation!
https://arduino-pico.readthedocs.io/en/latest/platformio.html#debug-port