earlephilhower/arduino-pico #1501: Use Serial1 with GPS

pierrotm777, 2023-06-05:
Hello,

Firstly, thank you for this nice adaptation for the RP2040.
I use a Waveshare RP2040 Zero for build a RC telemetry for a Radiolink transmitter.
For my GPS sensor, i need to use a Serial1 port on the pins 12 and 13.
I define my port like that:
#define MyGps Serial1

On setup():
MyGps.setRX(13);
MyGps.setTX(12);
MyGps.begin(9600);

But my program is blocked by this serial definition and i need to remplace it by a SofwareSerial fonction witch running very well.
The default pins for Serial1 are 0 and 1 but, normally, it is possible to use also the pins 12 and 13.
Any idea why i can't use Serial1 on these pins ?

Thanks for your help,

Pierre

pierrotm777, 2023-06-05:
I do that as soon as possible :-)

pierrotm777, 2023-06-05:
I send you my example for read gps values on the pins 12 and 13.
If i use  MyGps.setRX(13) and  MyGps.setTX(12) , the code don't see the gps.
If i use  MyGps.setRX(12) and  MyGps.setTX(13) , the code don't start and my pc return me that a  usb device isn't reconnized.

For SoftwareSerial i need to define:
static const int RXPin = 12, TXPin = 13;
SoftwareSerial MyGps(RXPin, TXPin);

[PRM_01_RP2040_TestSerial1.zip](https://github.com/earlephilhower/arduino-pico/files/11651456/PRM_01_RP2040_TestSerial1.zip)

pierrotm777, 2023-06-05:
Yes, I invert well Rx Tx and my GPS is well detected with SofwareSerial method.

LinusHeu, 2023-06-05:
> and my GPS is well detected with SofwareSerial method

Yes. But with SoftwareSerial you have RX and TX the other way around compared to Serial1 (Hardware Serial). This makes me think the issue could be the wiring.

If RX and TX are connected the wrong way around, it doesn't work.

Hardware: RX = 13, TX = 12. => doesn't work
Software:  RX = 12, TX = 13.  => RX and TX are swapped compared to Hardware => works.
=> I would try swapping the RX and TX wires around. And then try if the hardware method (with RX=13 and TX=12) works.

[image]

(I have seen some devices where RX and TX are labelled the wrong.)

LinusHeu, 2023-06-05:
I mean soldering or plugging the wires the other way around. But keeping your program the same. So, keeping setRX(13) and setTx(12).

pierrotm777, 2023-06-05:
I can't do that because, I use a specific board witch run already with these connections but use Pico sdk with VScode in this case. With Arduino code, I have an issue.

closed 2023-06-05 by earlephilhower