earlephilhower/arduino-pico #1507: SPI:  MSBFIRST/LSBFIRST and BigEndian/LittleEndian are independent settings!

board707, 2023-06-06:
Hi
There is some confusion between these two pairs of settings, but they are generally independent. The MSBFIRST/LSBFIRST controls the order of bits in a single byte, the BigEndian/LittleEndian - the order of bytes in a 32-bit word. Changing the order of bits when setting the LSBFIRST does not mean changing the order of bytes.
However, in your code these settings are treated as related. For example, in the `reverse16Bit` procedure, when changing from MSB to LSB, not only bits, but also bytes are unfolded:
```
inline uint16_t SPIClassRP2040::reverse16Bit(uint16_t w) {
    return (reverseByte(w & 0xff) << 8) | (reverseByte(w >> 8));
}
```

earlephilhower, 2023-06-06:
The Arduino docs are pretty high-level, so do you have any other core/MCU you use that does the MSB/LSB switch differently?  I'd like to just confirm, especially since this call is only used when users explicitly try and read 16 bits (which to me implies those 16-bits are a logical value and we should do magic to make it usable as a `short int` here in the core).  But, that's just my thinking in absence of other evidence...

Would you like to make a PR fixing this, since you found it (and I assume you ran into a problem because of it)?

board707, 2023-06-06:
Thank for your answer
I haven't run into a problem yet, however I'm writing code and I need to provide to users the ability to change LSB to MSB. And I'm not sure exactly how that should work.
As I wrote, there is some confusion in this matter. These two settings are described to be independent. For example, I work with the Winner Micro controller W806, which does not change the order of bits in bytes when changing the order of bytes from LITTLE to BIGENDIAN. But I understand that this can hardly be evidence.
I'm sorry, it would probably be more correct to transform this to discussion - but I don't see such an option. My post is not a complaint about your package, but rather a reflection on whether it works correctly.