earlephilhower/arduino-pico #1516: Proposed Linker Script and/or Menu Changes

PaintYourDragon, 2023-06-08:
Since this is your show, thought I’d ask your input on an approach before working through and submitting a PR you might not like…

First: there’s a file I’ll want to add to the default linker script that’s needed for deep sleep. (I’ll need to read up on this — don’t know the exact modification yet, but Limor tells me it’s a thing folks are doing.) I think it’s just one file, and shouldn’t affect anything size-wise if functions within are not invoked.

Second: there’s two files (I think it’s two…again, will double-check and get back to you) that absolutely must be loaded in RAM if using the PicoDVI library and writing flash memory (e.g. FAT filesystem). Two ways one might go about this…

1. Just do this in the default RP2040 linker script. Those functions, if invoked, will rob some RAM even if not using that specific DVI/flash combo…but…the items in question are small and fairly esoteric (e.g. interpolator peripheral save/restore), so this might not even affect most people. OR…
2. I noticed the boards.txt structure does have an element for specifying a linker script…but it’s currently ignored. So, second option here would be to have a menu selection for linker script A/B/C, and then add code to the build process to work with that selection. But given the low footprint of the material mentioned above, unsure if all this is really necessary and might be premature optimization.

In either case, I’ll do the work and submit a PR. Just wanted to get your input on a preferred direction.

Thanks!

earlephilhower, 2023-06-08:
Thanks for checking in.  I think it's all doable, but maybe less work than imagined.

For deep sleep, if you're pulling in the `pico_extras` deep sleep code, I would suggest not touching linker/libraries and just pulling in the single file and placing it in `cores/rp2040/sdkoverrides`.  Pulling in pico_extras was a real pain last I looked at it.

For files/functions in RAM, why not just annotate the important functions using the SDK decorator `__no_inline_not_in_flash_func()`.  You can also add a GCC `.section` prefix to the file  (in an ASM block) which could also place everything in RAM in a compilation unit.  We did this for some ESP8266 work, IIRC...

PaintYourDragon, 2023-06-08:
Ah ha! Wasn’t aware of sdkoverrides, I’ll look into that a bit. Thanks!

I had started down the __not_in_flash_func() path but there were just layers and layers of function calls that I couldn’t track them all down…and it’s super time-consuming because getting it wrong crashes the board and wipes out the filesystem. What had been done in CircuitPython (where the same problem was dealt with) was a tweak to the .ld file to put a couple core library files and the PicoDVI lib in RAM. Thus it was all centralized in that one file and didn’t require changes throughout both the core and library.

That was all before the weekend, so excuse me that it’s not all fresh, but give me a sec to at least dig up that CircuitPython linker file, maybe it gives some insights.

PaintYourDragon, 2023-06-08:
https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/link.ld

Relevant bits seem to be lines 143 and 83.

My next step is to simply try these changes in the arduino-pico ld script and see if that fixes our issue. A bit under the weather so I’ll have to come back to this, perhaps tomorrow. Thanks for your patience with this.

earlephilhower, 2023-06-08:
No worries, and thanks for the link.  Looks like just a couple add'l support functions (divider) and one uniquely named file is being added.

I think for this, no need to make new .LD file and worry about infrastructure.

If you do want a new LD file I would suggest we fix the custom-LD-per-board in makeboards.py and platform.txt, instead of making a new menu item because this would be the only board using it and it would also be one more thing users could get wrong w/the PicoDVI.

earlephilhower, 2023-06-09:
I did a quick test and with the LD changes I'm not seeing any difference in RAM usage in sketches I've tried.  I think the add'l divider.o might already have functions in RAM using the decorator so it's redundant.  (We don't use the interpolater so that's a no-op change normally).

So, when you're feeling better just update the main LD and save your efforts.  Good luck!

PaintYourDragon, 2023-06-15:
Thank you for your patience…fell into a huge rabbit hole of tangent code. ANYWAY, a PR has been opened, impact is minimal such that this can be done in the default linker script, no need for A/B select yet. Gonna close the issue and any further discussion if needed can be done in PR. Thx! 🐲

closed 2023-06-15 by PaintYourDragon