earlephilhower/arduino-pico #1517: To many incompatibilities sdfat <> ESP8266SdFat?

madias123, 2023-06-09:
Hello,
sadly there are too many conflicts with the ESP8266SdFat library especially with other libs (like Adafruit ones...)
One simple example:
```
#include <SdFat.h>
SdFat sd;
File file;
char fileName[] = "datalog.csv";
#define LOGFILE "datalog.csv"

void setup() {
  file = sd.open(fileName, FILE_WRITE);
}
void loop() {
  // put your main code here, to run repeatedly:
}
```

results in
` error: 'File' does not name a type; did you mean 'SdFile'?`
taken from here (issue was fixed on teensy):  [https://github.com/greiman/SdFat/issues/291](url)

there are many other issues like
`error: could not convert 'FatVolume::open(const char*, oflag_t)(((const char*)"/"), 0)' from 'File32' to 'fs::File'`

meanwhile the "customer" is confronted with at least 3 versions of sdfat:
old one by greiman
V2 by greiman
ESP8266SdFat
Adafruit Sdfat
teensy one (Stoffregen is doing his best to stay compatible)
_Sadly there is no way to guarantee cross compatibility using  the sdfat library (and the original one ist outdated since 2009)_

earlephilhower, 2023-06-09:
It's incompatible because we already have a `File` class, internally, for LittleFS.

In any case, there is no need to use SDFAT.  Please use `#include <SD.h`> or `#include <SDFS.h>`  to use the built-in wrapper that maps the SdFat stuff to our internal `File` operations.

closed 2023-06-09 by earlephilhower

madias123, 2023-06-09:
Thank you, I think, I understand the dependences for now ;) 
Regards Matthias