earlephilhower/arduino-pico #1532: New board isn't recognised?

doctea, 2023-06-13:
Apologies if this is (another) dumb question.  I'm trying to add a new board, and I think I've followed the instructions on the 'Contributing' page.  Everything looks in place as I understand it, but when I attempt to use the board by setting it in platform.ini, I get `Error: Unknown board ID 'pimoroni_plasma2040'`.

I've uploaded my modified changes here: https://github.com/doctea/arduino-pico/tree/pimoroni_plasma

Is there a step I'm missing?

earlephilhower, 2023-06-14:
Sorry, I don't use Platform.IO so can't be of much help.  My wild guess would be to double check you are pointing to your correct github URL and branch name for your new fork and branch.

maxgerhardt, 2023-06-15:
If you're adding a new board do Arduino-Pico and want to use it in PlatformIO, you must also add the generated board JSON file to the platform-raspberrypi's `boards/` folder. See https://github.com/maxgerhardt/platform-raspberrypi/tree/develop/boards.

Usually after a board PR has landed I'll update the available boards in the platform myself, but you can emulate by copying it in your platform installation too. (Windows: `C:\Users\<user>\.platformio\platforms\raspberrypi<@src-....>\boards`)

A quicker alternative is to create a `boards`  folder locally in your project and copy the `pimoroni_plasma2040.json` file in there.

earlephilhower, 2023-06-16:
@doctea I think we're done here.  If so, could you close this?

closed 2023-06-17 by earlephilhower

doctea, 2023-06-22:
Apologies for the delay, I've just confirmed the steps given work (adding a local copy of the .json file to a 'boards' folder in the project folder).  Thanks for your help @maxgerhardt !