earlephilhower/arduino-pico #1533: Problems when running large buffers using over half the available ram.

XanCraft21, 2023-06-13:
Board: Raspberry pi pico, Adafruit qt py RP2040

Package version: (i need to list many versions as this happened about a few versions ago, but i don’t remember when it started. I will list them later.)

Platform: Arduino
Operating system: Windows 10

Hello, i absolutely love arduino package, it has made working with the pico so easy. But i have encountered a problem with some recent updates. I no longer have full ram access when working with large data buffers.

When working with buffers for lcd displays i used to be able to get a size of up to 240x320 and still have a quarter of the ram leftover, of course using 16 bit graphics. But when i do it now i can barely make a 240x240 buffer, working best at 135x240.

This problem may only be on my end, but i want to be sure. If you are able to look into this and find a solution i will gladly appreciate it. Thank you for your help.

earlephilhower, 2023-06-13:
Sorry, you'll need to provide a simple MCVE that doesn't need any special HW or else there's not much that can be done.

As a guess, I think you might be hitting memory fragmentation in your code.  So try and allocate the large screen buffer right at the beginning of `setup` so that it can grab a large contiguous chunk of RAM.  OTW, you could have, say, 200KB free but in 4 x 50KB chunks separated by a few allocated bytes.

XanCraft21, 2023-06-14:
It may be very hard for me to get a MCVE. I will need some help doing that.

I will try the memory allocation test when i can, but it might take me a few days since i will be away from home for about a week. I will get back with you when i can.

earlephilhower, 2023-06-14:
Another possibility would be to allocate a dummy block larger than you need for the display at the beginning of setup, then just before you start the library code to really allocate `free()` the block so the memory manager will be able to use that contiguous chunk.

earlephilhower, 2023-06-17:
When run w/the latest core I can allocate 200KB right at program start:

````
void setup() {
  delay(5000);
  Serial.printf("malloc 200KB = %p\n", malloc(200*1024));
}
void loop() {
}
````

returns
````
malloc 200KB = 0x20002830
````

So, there's nothing wrong with large mallocs as long as you have a large enough contiguous block.  Since there's no MMU that's the best we can do on this chip.  Closing.

closed 2023-06-17 by earlephilhower

XanCraft21, 2023-06-19:
As i said before it may be a problem on my end. I might need a new computer. Thank you for checking this for me.

Another thing could be that the library i was using is improperly creating the ram buffer which most libraries for lcd displays do, but i will still go with my pc having corrupted and unfixable software.

Again, thank you for checking this for me.