earlephilhower/arduino-pico #1534: Problems when updating the board version in Arduino

XanCraft21, 2023-06-13:
Board: Raspberry pi pico, Adafruit qt py RP2040

Package version: (this problem has been happening for a while now, i will need to dig to find the version this started happening with.)

Platform: Arduino
Operating system: Windows 10

Hello, i really love this boards package, but there is a problem when the package has an update.

The problem goes as follows, once i start the update it deletes the old files then begins installing the news ones as normal, but it gives an error saying a certain file already exists, when it was literally just installed.

If i close Arduino and try again it finishes but doesn’t install the other tools, but it installs all the boards. When i compile it gives an error saying it’s missing a python file named “Encodings”.

What i have to do is uninstall the boards package, delete the folder inside the packages folder, inside the Arduino folder, inside appdata. Then i need to delete the json file relating to this repo, then if i install it again it goes fine. I need to do this every time i need to update the package.

It may be a problem on my end caused by computer hackers, but i am unsure. If you can find a way to fix this i will gladly appreciate it. Thank you for your help.

earlephilhower, 2023-06-13:
This is an Arduino IDE problem.  See https://github.com/earlephilhower/arduino-pico/issues/976#issuecomment-1312727031 for a solution.

closed 2023-06-13 by earlephilhower

XanCraft21, 2023-06-14:
So i guess that means there will be no fix. Thank you for your reply.

earlephilhower, 2023-06-14:
We have no control over how the IDE does the installation.   V2.0 of the IDE has not had this problem reported (but V2 doesn't allow plug ins so the LittleFS upload isn't supported...(

XanCraft21, 2023-08-29:
I have found a solution. It took me a while to figure out since no-one wanted to help me.

I had to remove the boards package, then find the appdata folder, Arduino, Packages, RP2040, go into tools then there is a folder that didn’t get deleted inside the tools folder.

I have to delete the python folder inside the tools folder, then the package reinstalled fine. I hope this only needed to be done once.

tesla-ES, 2024-05-11:
> I have found a solution. It took me a while to figure out since no-one wanted to help me.
>
> I had to remove the boards package, then find the appdata folder, Arduino, Packages, RP2040, go into tools [...]

it's work nice with me too