earlephilhower/arduino-pico #1540: Error Downloading on Work Computer

Sir-Laggzalot, 2023-06-15:
I'm having an issue downloading the board manager through Arduino. I'm using a work computer, but I was able to download the Teensy board manager on this computer, so not sure why this is happening for this one.

[ArduinoRP2040ErrorMessages.txt](https://github.com/earlephilhower/arduino-pico/files/11761173/ArduinoRP2040ErrorMessages.txt)

Sir-Laggzalot, 2023-06-15:
I was able to install it through Git, so no big deal if there is no known solution for this. Leaving it open in case you want to investigate it, but feel free to close it.

earlephilhower, 2023-06-15:
Looks like your work computer has something odd with Java or its certificate store (maybe locked down hard).

> Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

Nothing we can do here, so closing.  Glad to hear you got it going eventually.

closed 2023-06-15 by earlephilhower