earlephilhower/arduino-pico #1545: Serial + I2S = only noise

LinusHeu, 2023-06-17:
Commenting the Serial.begin()... => it works.
Having it uncommented => only loud noise is played.
Removing the entire loop() content: only silence. No matter if with or without Serial.

Of course, adding Serial.print() in the playback-loop can cause too much delay and then underflows.
Because the Serial stuff is outside the playback-loop I expect it to have no effect on audio. 
```ino
#include <Arduino.h>
#include <I2S.h>
#include <StarWars30.h>

long sampleRate = 22050;
I2S i2s = I2S(OUTPUT);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  i2s.setBCLK(26);  // WS = 27
  i2s.setDATA(28);
  i2s.setBitsPerSample(16);
  i2s.setBuffers(4, 64);
  //=====commenting this => playback works
  Serial.begin();
  while (!Serial);
  delay(1000);
  Serial.println("Starting");
  //=====
  if (!i2s.begin(sampleRate)) {
    digitalWrite(LED_BUILTIN, HIGH);
    while (true);
  }
}

void loop() {
  const unsigned char *ptr = StarWars30_raw;
  unsigned int toPlay = StarWars30_raw_len;
  while (toPlay > 10) {
    //===Make the track stereo and quieter
    int16_t frame[2];
    frame[0] = (*(int16_t *)ptr) / 16;
    frame[1] = (*(int16_t *)ptr) / 16;
    //===Write one L+R sample to I2S
    if (!i2s.write16(frame[0], frame[1])) {
      digitalWrite(LED_BUILTIN, HIGH);
    }
    toPlay -= 2;
    ptr += 2;
  }
  //===END
  i2s.end();
}
```

LinusHeu, 2023-06-17:
This following sketch uses the [arduino-audio-tools library](https://github.com/pschatzmann/arduino-audio-tools). And it works. Even with a Serial.println("Starting").

```ino
/**
#include "AudioTools.h"
#include "StarWars30.h"

AudioInfo info(22050, 1, 16);
I2SStream i2s;  // Output to I2S
MemoryStream music(StarWars30_raw, StarWars30_raw_len);
StreamCopy copier(i2s, music); // copies sound into i2s

void setup(){
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN,LOW);
    Serial.begin(115200);
    while(!Serial);
    delay(5000);
    Serial.println("Starting");
    digitalWrite(LED_BUILTIN,HIGH);

    AudioLogger::instance().begin(Serial, AudioLogger::Warning);

    auto config = i2s.defaultConfig(TX_MODE);
    config.copyFrom(info);
    if(!i2s.begin(config)){
      while(1){
        Serial.println("FAIL");
      }
    }

}

void loop(){
    if (!copier.copy()){
      i2s.end();
      digitalWrite(LED_BUILTIN, LOW);
      stop();
    }
}
```

platformio.ini:
```
[env:rpipicow]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipicow
framework = arduino
board_build.core = earlephilhower
monitor_echo = true
;lib_deps = https://github.com/pschatzmann/arduino-audio-tools
build_flags = -D USE_I2S -DCORE_DEBUG_LEVEL=2 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args 
monitor_filters = log2file
platform_packages = framework-arduinopico@symlink://C:\Users\linus\Projekte\arduino-pico
```

LinusHeu, 2023-06-17:
The length of the string that gets printed causes the noise!

With my initial sketch that doesn't use the arduino-audio-tools library:
Serial.println("s");
Serial.println("S");
Serial.println("st");
Serial.println("sta");
Serial.println("123");
=> work 

Serial.println("star");
Serial.println("starting");
Serial.println("STARTING");
=> cause loud noise.

earlephilhower, 2023-06-17:
Wow, that is odd.  I can reproduce it here on my Adafruit maxXXX DAC setup.

If I add a `delay(5000)` after the serial.print I get silence for those 5 seconds then the noise.  This indicates to me that somehow the data shifted out is being misaligned somewhere (since silence is all 0s, shifting it left or right won't cause any change in the actual read value by the DAC).

How `Serial` could make that happen, I can't imagine yet...

earlephilhower, 2023-06-17:
Looks like this is a bug in your code, not the core.  You're doing an unsafe cast from `char *` to `int16_t*` (gcc warns about it, too, if you have all warnings enabled).  There are alignment requirements on this chip, so if the music array starts at an odd address (legal for char, not legal for int), bad things will happen.

I just changed your array to align on a 4 byte boundary (just for extra safety) and I can print lots of stuff w/o any issue:
````
const unsigned char StarWars30_raw[]   __attribute__ ((aligned (4))) = {
.....
````

LinusHeu, 2023-06-17:
That fixes the issue. Thank you so much for all your help. I really need to be more careful with typecasts. :D 

Just one more question, if you have the patience:
char/uint8_t arrays in RAM are also NOT guaranteed to be aligned on a 2/4 byte boundary, right?
```cpp
uint8_t *array = new uint8_t[1024];
int16_t oneSample = *(int16_t*)array;
```
is equally unsafe as what I did with the array in memory, correct?
Now I'm worried that other code I've written only works by chance.

closed 2023-06-17 by LinusHeu

earlephilhower, 2023-06-17:
I believe you will get 4 byte aligned addresses from `new` or `malloc`, so I would not worry about those.  Only the `char` arrays that the linker will try and squeeze together should be of concern.

LinusHeu, 2023-06-17:
Great! Thanks again.

LinusHeu, 2023-06-21:
> gcc warns about it, too, if you have all warnings enabled

What warnings do you have enabled? Or recommend I enable?
For no reason, I assumed that -Wall would be on by default. :D 
Now I added `build_flags = -Wall -Wextra` in my platformio.ini and I still don't get any warnings with that code. (I confirmed in general I can trigger warnings. For example, adding an unused variable creates a warning.) Same in the Arduino IDE 2.1.0 with "Compiler warnings: all".

earlephilhower, 2023-06-22:
In the 1.8.x IDE with all warnings I get a `type-punning` warning when you dereference the (char *) as (int16_t) in
````
    frame[0] = (*(int16_t *)ptr) / 16;
    frame[1] = (*(int16_t *)ptr) / 16;
````

It may just be that you ran into a C [footgun](https://en.wiktionary.org/wiki/footgun)!

LinusHeu, 2023-06-22:
Just tested in the 1.8 IDE and no warnings there either.
I'm fine with keeping it like this. :D But just in case you think there might be something wrong going on with the build-process, toolchain, etc, here's the output of the Arduino IDE 2.1 with compiler warnings set to all and verbose output during compile.
Btw, I'm on Windows 11.

[ArduinoIDE_2-1-0_verboseBuild_WarningsAll.txt](https://github.com/earlephilhower/arduino-pico/files/11832837/ArduinoIDE_2-1-0_verboseBuild_WarningsAll.txt)