earlephilhower/arduino-pico #1546: PlatformIO recursively clones the entire PicoSDK module, with its own submodules.

ProtonNumber, 2023-06-18:
Installing this module on platformIO causes it to clone the PicoSDK, and all submodules. 
This includes TinyUSB, which has submodules for every hal it supports.

This is a really minor issue, as it doesnt impair the functioning of the library, but it makes installation take forever.

earlephilhower, 2023-06-18:
As we use the Pico-SDK unmodified, there's nothing to be done here about it.  The next release will have the SDK 1.5.1  which will shrink the TinyUSB install significantly.  You can (should) use the `git` version and not the Platform package as P.IO has disallowed me from publishing new versions for a couple months because they're removing all Raspberry Pi Pico support soon.  If you use `git now, you'll get that smaller install since the 1.5.1 SDK was merged here yesterday.

closed 2023-06-18 by earlephilhower

ProtonNumber, 2023-06-19:
Wait, they're planning to remove pico support?

Thats a shame.

minisbett, 2023-06-25:
What exactly do you mean by "remove pico support"?

earlephilhower, 2023-06-25:
https://github.com/platformio/platform-raspberrypi/pull/36

minisbett, 2023-06-25:
> [platformio/platform-raspberrypi#36](https://github.com/platformio/platform-raspberrypi/pull/36)

I read it but couldn't really understand the issue. PIO is trying to force RPi Trading to pay that they explicitly said they have no interest?

earlephilhower, 2023-06-25:
https://github.com/platformio/platform-raspberrypi/pull/36#issuecomment-1566052525

Ing-Dom, 2023-07-04:
> You can (should) use the `git` version and not the Platform package as P.IO has disallowed me [...]
what exactly do you mean with that and how do I do it?

Is it this what's currently marked "deprecated" ?
```
platform_packages =
maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/
```