earlephilhower/arduino-pico #1552: Incorrect SD card size and usedBytes > totalBytes reported by SDFS

mef51, 2023-06-20:
Hello, I'm using a 16 GB micro SD card with the Pico W and the size is being incorrectly reported with `SDFS.info64` and `FSInfo64`

Example sketch:
```cpp
#include <SDFS.h>
#define CS_PIN 17 // Chip Select (SPI)
#define CD_PIN 22 // Card Detect
FSInfo64 SDinfo;
SDFSConfig sdconf;

void setup() {
  sdconf.setCSPin(CS_PIN);
  SDFS.setConfig(sdconf);
  SDFS.begin();
  SDFS.info64(SDinfo);
  Serial.print("total bytes: "); Serial.println(SDinfo.totalBytes);
  Serial.print("used bytes: "); Serial.println(SDinfo.usedBytes);
}

void loop() {}
```

Output:
```
total bytes: 3038248960
used bytes: 4295098368
```
That's 3.03 GB total and 4.3 GB used bytes. 

I can write to the card without issue using SDFS and plugging the card into my desktop reports the correct size and files

earlephilhower, 2023-06-20:
Please try (the double 'll's are important to ensure printf uses 64-bits)
`Serial.printf("Total Bytes:  %lld\n", SDinfo.totalBytes);`

I don't think `Serial.println()` handles 64-bit quantities properly.  That would be an Arduino API limitation and not something we can do much about here...

mef51, 2023-06-20:
I get the same output
```python
void setup() {
  sdconf.setCSPin(CS_PIN);
  SDFS.setConfig(sdconf);
  SDFS.begin();
  SDFS.info64(SDinfo);
  Serial.printf("Total Bytes:  %lld\n", SDinfo.totalBytes);
  Serial.printf("Used Bytes:  %lld\n", SDinfo.usedBytes);
}
```
```
Total Bytes:  3038248960
Used Bytes:  4295229440
```

I tried some other ways of printing 64 bit numbers from [SO](https://stackoverflow.com/questions/36188155/how-do-i-output-a-64bit-number-on-arduino-if-its-libraries-have-32-bit-restricti) but got the same number

earlephilhower, 2023-06-20:
Thanks for checking.  I'll have to wait until the weekend to hook up a card and see what's going on.

earlephilhower, 2023-06-21:
@mef51 could you please give #1553 a try?  I haven't got the HW hooked up, but on code inspection it looks like I was doing a 32-bit multiply instead of a 64-bit one to get the total card size, so it would overflow.  Just added some typecasting to make sure that doesn't happen.

mef51, 2023-06-21:
Sure I can try this out at the end of the week, thanks

PR #1553 (2023-06-23): Ensure 64bit math for SD card FSInfo
Fixes #1552

Ensure that 64 bit multiplication is done when calculating the total size available and used for SDFS.
commit 9e89dda9 (2023-06-23): Ensure 64bit math for SD card FSInfo (#1553)

Fixes #1552

Ensure that 64 bit multiplication is done when calculating the total size
available and used for SDFS.
closed 2023-06-23 by earlephilhower