earlephilhower/arduino-pico #1561: `idleOtherCore` causing the microcontroller to die

minisbett, 2023-06-25:
I have written a firmware that can receive a `save` command via the serial interface. Upon receiving, it puts a configuration struct into the EEPROM and commits it. But after committing, the whole micro controller dies.

To recall what `EEPROM.commit()` does:
```cpp
noInterrupts();
rp2040.idleOtherCore();
flash_range_erase((intptr_t)_sector - (intptr_t)XIP_BASE, 4096);
flash_range_program((intptr_t)_sector - (intptr_t)XIP_BASE, _data, _size);
rp2040.resumeOtherCore();
interrupts();
```

This only happens if I use both cores. If I don't have `setup1` and `loop1` specified, everything works as intended. I looked up the code for `idleOtherCore` and tested out what the exact issue is, which turned out to be the idle of the other core.
What's worth mentioning is that the currently executing method does run until the very end.
Example:
```cpp
void func1()
{
    EEPROM.commit();
    Serial.println("foo");
    Serial.println("bar");
}

void func2()
{
    func1();
    Serial.println("foobar"); // NOT being executed
```

I have tried to mess around a lot with the arduino-pico code to figure out what the issue might be, without success.
I tried running an empty project, just testing the `idleOtherCore` method and this worked without any issues:
```cpp
void setup() { Serial.begin(115200); delay(5000); }

void loop()
{
  Serial.println("Idling core1 for 3 seconds");
  rp2040.idleOtherCore();
  delay(3000);
  Serial.println("Resuming core1");
  rp2040.resumeOtherCore();
  delay(1000000);
}

void loop1() { Serial.println(millis()); }
```

Then I tried setting up something similar in my firmware which did cause the issues again, but very inconsistently:
```cpp
void loop1() { Serial.println(millis()); }

#pragma endregion

void serialEvent()
{
    Serial.println("Idling core1 for 3 seconds");
    rp2040.idleOtherCore();
    delay(3000);
    Serial.println("Resuming core1");
    rp2040.resumeOtherCore();
    
    // Handle incoming serial data.
    String str = Serial.readStringUntil('\n');
    SerialHandler.handleSerialInput(&str);
}
```
This causes the microcontroller to freeze *sometimes*, depending on my serial input. If I send something like `echo foobar`, a command in my firmware for debugging purposes, everything works fine. (Except for the `readStringUntil` taking **exactly** 1 second if ran after the idle & resume stuff)
But if I run a command like `get`, which outputs my entire config, the MCU freezes/dies in the middle of printing it out. This again does not happen if I am not using the 2nd core.

In conclusion, my assumption is that there is an issue with using the `idleOtherCore` function in combination with serial actions, although the `EEPROM.commit()` issue happens **after** my whole command handling, unlike the test code above which idles the core before doing my serial stuff.

earlephilhower, 2023-06-25:
`idleOtherCore` is not recursive.  It's a very simple high prio IRQ to get the opposite core running from SRAM until it gets an "all clear" from the initiating core.

Once a core is asleep the intention is to do something fast and dangerous (i.e. writing to flash) and then resume the other core.  `delay()` may work while locked, but no guaranteed (especially w/FreeRTOS where you are guaranteeing a hang).

Why are you trying to stop the other core?  You can use a `mutex_t` (or any other standard exclusion stuff) to protect access to some shared global or resource, for example.  Then the 2nd core never *has* to stop doing whatever it's doing, only when there would be a conflict.

In your last example, the problem is your code in `loop1` grabs the SerialUSB mutex and starts `println`, then gets frozen **while holding that mutex**.  Then core0 tries to grab the mutex but, since core1 has it and is frozen the mutex will never be freed.  Deadlock.  I haven't gone thru your other examples, but I bet that they'd all boil down to either this deadlock or calling `idleOtherCore` recursively.

closed 2023-06-25 by earlephilhower

minisbett, 2023-06-25:
> `idleOtherCore` is not recursive. It's a very simple high prio IRQ to get the opposite core running from SRAM until it gets an "all clear" from the initiating core.
>
> Once a core is asleep the [...]

I don't think you really understood the issue. `EEPROM.commit()` does not properly because it idles the other core.

earlephilhower, 2023-06-25:
> I don't think you really understood the issue. EEPROM.commit() does not properly because it idles the other core.

That's exactly what it has to do, stop the other core from running for a bit.  When flash is erased/written you can't read from it, so the other core that's not writing flash needs to be idled in RAM.  If you don't, then the XIP cache can try and read while the flash is busy and you'll have a HW fault.

minisbett, 2023-06-25:
But why does the micro controller stop running any core once I ran `EEPROM.commit`? It continues to fully run the function the commit function is called from but then dies.

earlephilhower, 2023-06-25:
Lots of people are using EEPROM (although `LittleFS` would be a better choice).  While it's possible there's an issue, your posted code snippets look to all have deadlocked mutexes.

I suggest you hook up a SWD debugger and dump both core's stack traces.  I bet you'll find the deadlock I mentioned.

minisbett, 2023-06-25:
> Lots of people are using EEPROM (although `LittleFS` would be a better choice). While it's possible there's an issue, your posted code snippets look to all have deadlocked mutexes.
> 
> I suggest you hook up a SWD debugger and dump both core's stack traces. I bet you'll find the deadlock I mentioned.

All except for the second code snippet are just test snippets. My actual application literally only calls `EEPROM.commit`, which causes both cores to crash completely. I don't think I'm using the commit method improperly here. I'll try to set up a debugger but don't have the required hardware necessary available right now.