earlephilhower/arduino-pico #1568: Disable COM with USBCore

0xN4TH4N, 2023-06-28:
Hello, I want to use this code (https://gist.github.com/0xN4TH4N/359bc189ae083d6ecd76b5de95e05e81) created for arduino, who will disable COM. I want to use this for arduino pico. But it does not work. Can someone help me ?

earlephilhower, 2023-06-28:
That is code for an 8-bit AVR and completely incompatible with anything else.  To disable USB, just use `Tools->USB Stack->No USB` in the IDE menus.

closed 2023-06-28 by earlephilhower

earlephilhower, 2023-06-28:
That code is useless on any non 8-bit AVR board, so for sanity I'd just ignore it if I were you.  There's no correspondence between the special HW registers it's twiddling and the ones on this chip (or any other chip).

You cannot disable `Serial` with the default USB stack, but you can use `Tools->USB Stack->Adafruit` and then use their code (see the examples) to make a gamepad-only HID device.