earlephilhower/arduino-pico #1569: Wifi or pico crash after a moment

0xN4TH4N, 2023-06-28:
I am encountering an issue with my Pico W microcontroller where it successfully handles socket connections initially, but after a certain point, the entire server crashes without any apparent reason or error message. This occurs consistently and has been observed after receiving approximately 26,000 socket connections.

I have developed a server application using the Pico W microcontroller, which is responsible for accepting and processing socket connections from various client devices. Initially, the server operates smoothly, successfully handling a large number of socket connections. However, after reaching around 26,000 socket connections on average, the entire server abruptly crashes, causing disruption to the connected clients and rendering the server inaccessible.

Server loop
```
void loop() {
  delta[0] = 0;
  delta[1] = 0;
  delta[2] = 0;

  WiFiClient client = server.available(); 
  //client.setNoDelay(true);
 
  if (client) {
    while (client.connected()) {
      if (client.available() > 0) {
        char inChar = client.read();
        if (inChar == 'M') {
          client.readBytes((char *)&delta, 2);
        }
        client.flush();
        # TODO : Receive
      }
    }
    client.stop();
  }
}
```

Client side
```
self.socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
self.socket.setsockopt(socket.IPPROTO_TCP, socket.TCP_NODELAY, 1)
self.socket.connect((config.get('CONFIG', 'IP'), self.port))

def move(self, x, y):
  self.socket.send(b"M" + bytes([int(x), int(y)]))
```

I am willing to provide further details, share specific sections of the server code, or conduct additional tests as requested. Your support and expertise in resolving this issue would be invaluable. Thank you for your attention and assistance.

earlephilhower, 2023-06-28:
We need a small (1-screen's worth) and complete self contained example that shows the problem to do anything here.  Something we an just cut and paste into the IDE, update the WiFi credentials, and go.

If you can't supply that, then I would suggest hookup up a SWD connector and using GDB to see what exactly your code is doing when the crash happens.

0xN4TH4N, 2023-06-28:
I have nothing even by activating all the debugs, the only thing is a closure that I see on the client side 
```
self.socket.send(b"M" + bytes([int(x), int(y)]))
ConnectionResetError: [WinError 10054] An existing connection had to be closed by the remote host
```

earlephilhower, 2023-06-28:
If this is your mouse w/o Serial port like in #1568 you're not going to get any output, debug or otherwise, of course.

In any case, I would suggest making that MCVE as requested, verifying it still fails, and posting it.  Or try the SWD debugging which is very powerful and easy with this chip.

0xN4TH4N, 2023-06-29:
I can't debug like this, does there is a another method ?

maxgerhardt, 2023-06-29:
There's probably a memory leak in releasing the resources associated with the accepted WiFi client in either your code or the core / library code. There are functions to track heap usage.

https://github.com/earlephilhower/arduino-pico/blob/3778fbb833918c9a09433f85785456ec074edb22/cores/rp2040/RP2040Support.h#L227-L239

Can you just add this in every loop iteration? 
```
Serial.println("Free heap: " + String(rp2040.getFreeHeap()) + " Total: " + String(rp2040.getTotalHeap()));
```
if this goes **down** with every connection, that's a smoking gun.

0xN4TH4N, 2023-06-29:
It works very well, same, after a few minutes, it stops and does not turn the loop. I feel like the temperature plays a role. I can touch that the more time passes, the more the temperature increases.

```
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
Free heap: 189688 Total: 190000
```

maxgerhardt, 2023-06-29:
Then it might get stuck in the underlying WiFi stack on either the Pico (receiving side) or worse, the Cypress chip.

Do you have a second Pico board to facilitate picoprobe /CMSIS-DAP debugging per [documentation](https://arduino-pico.readthedocs.io/en/latest/platformio.html#debugging)?

0xN4TH4N, 2023-06-29:
No, I don't have 2 pico boards

earlephilhower, 2023-06-29:
If you're somehow cooking the chip, then one thing to try is just `rp2040.reboot()` every 15,000 connections.  If it is temp/power related then it should still die after ~30K connections since reboot only takes a few seconds.  If it keeps running indefinitely well beyond 30K connections you probably have some other issue.

I'm going to close this for now.  If you can give a complete failing MCVE we can re-open.

closed 2023-06-29 by earlephilhower