earlephilhower/arduino-pico #1571: can't use Mouse.click function

Veeeeeski, 2023-07-01:
every time i use Mouse.click function after my Mouse.move function, the mouse.click function doesn't work. But when i reverse the functions it works (causing some issue). I would like to know if there is an alternative to perform a Mouse.click. Here is my codes :

     `delta[0] = bf[0];
      delta[1] = bf[1];
      delta[2] = bf[2];
    Mouse.move(delta[0], delta[1], 0);
         if(delta[2] == 1){
    Mouse.click(MOUSE_MIDDLE);
        }`

earlephilhower, 2023-07-02:
You're losing one of the USB messages because they're coming in too close tohgether with your snippet.  Try adding a `delay` between the move and click, or look at the `Mouse.press/release` calls.

closed 2023-07-02 by earlephilhower

Veeeeeski, 2023-07-02:
I tried using delay and i can use mouse.click but 50% of the time it's skipping some clicks. I was using this code on my arduino leonardo and it was working well even if Mouse.click and Mouse.move where too close together. So what can i do to avoid clicks getting skipped ?