earlephilhower/arduino-pico #1575: Feature request: Add USBtmc class

caizikun, 2023-07-04:
TinyUSB supports Test and Measurement Class (USBTMC) class. But arduino-pico have no related code. 
May I know usbtmc class would be added or not in feature?

lyusupov, 2023-07-04:
1. https://github.com/earlephilhower/arduino-pico/discussions
2. https://github.com/earlephilhower/arduino-pico/tree/master/libraries

earlephilhower, 2023-07-04:
Neat, I didn't know GPIB was still alive but in USB form!

Unfortunately it's something that very few users would have a need for and something for which I have no background or capability of testing.  You might want to ask in the Adafruit_TinyUSB repo if they're interested, or offer them a PR.  We support their USB stack as well so if it's in there, it's in here.

closed 2023-07-04 by earlephilhower