earlephilhower/arduino-pico #1580: HID keyboard with BIOS support

silpakcx, 2023-07-06:
Hi
I am trying to enable HID keyboard with Rp2040 for my project, but I couldn't find BIOS support with the latest library. Will the keyboard.h library supports BIOS page operations? If not is there any way to enable it.

earlephilhower, 2023-07-06:
It's not enabled now, but I believe all you;'d need to do is adjust the HID descriptor to indicate boot mode protocol support.  Check the RP2040USB.cpp file for where the USB descriptors are generated.

silpakcx, 2023-07-07:
okay I will try... thank you. In the next release plz try to include this feature too...

earlephilhower, 2023-07-07:
Always happy to get a PR for things like this, so if you get it working do please submit one!

closed 2023-07-07 by earlephilhower