earlephilhower/arduino-pico #1583: Fresh / Updated Install Failing with "board_build.core" option per docs

mawildoer, 2023-07-08:
G'day all! Firstly, thank-you for your great work! 

After coming back to my machine after lunch, I pulled my project's partner's work and poof, couldn't build anymore:

`UnknownPackageError: Could not find the package with 'earlephilhower/framework-arduinopico @ ~1.20303.0' requirements for your system 'darwin_x86_64'` -- Note: I've got an ARM Mac

I'm working in `platformio` with the following build config:

```ini
[env:earlhower_rp2040]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
; board_build.core = earlephilhower
```

Commenting out the last line finally meant I was once again able to build successfully.

Can I lend a hand tracking down what's going on here?

earlephilhower, 2023-07-09:
I'm not a P.IO user, but it looks like your problem is you're trying to use the framework from P.IO and not GIT.  P.IO has disabled publishing packages 

Please double check you're using the proper config per the docs https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst.  You may need to delete and reinstall the p.io support from Max from `git` .

maxgerhardt, 2023-07-17:
>UnknownPackageError: Could not find the package with 'earlephilhower/framework-arduinopico @ ~1.20303.0' requirements for your system 'darwin_x86_64' -- Note: I've got an ARM Mac

I do not reference that package version in https://github.com/maxgerhardt/platform-raspberrypi/blob/f9e5a07556f5a89b799bd01c377ba2ac342ab5a9/platform.json#L46-L51 in the current version. Please remove `C:\Users\<user>\.platformio\platforms\raspberrypi*` and press build again (will automatically redownload).