earlephilhower/arduino-pico #1584: Possible issue with Wire library (device: CY8C9520A port expander)

pmmarquez, 2023-07-10:
Hello,

The issue has to do `_probe` method of Wire.cpp when CY8C9520A is present, for example running Arduino i2c_scanner example.

Quick fix on `endTransmission`:
```cpp
    if (!_buffLen) {
        // Special-case 0-len writes which are used for I2C probing
        int ret;
        uint8_t rx_dummy;
        ret = i2c_read_blocking(_i2c, _addr, &rx_dummy, 1, false);
        return ret < 0 ? 2 : 0;
        // return _probe(_addr, _sda, _scl, _clkHz) ? 0 : 2;
    }
```` 
Maybe this device (CY8C9520A) has some issue with the timing of the bit-banging used on `_probe` or with the transmission of only the 8 bits of the address, with more time I will investigate deeper.

While trying to find a fix, I saw that Arduino mbed core does something similar to the quick fix I made, so I also have a question to gain more knowledge on the subject: why using this bit banging approach?

Perhaps a future solution will be to use some define to choose one or the other way.

Best Regards,

earlephilhower, 2023-07-10:
The bit-banging was because the HW cannot do a 0-byte transfer (it just won't start).  Per #38 and the Adafruit folks, a 0-len write is the normal way of doing this, not a read (AIUI a 1-byte read can potentially have side effects).

So I'm not so sure about the suggested change, especially since the current 0-len write seems to be working w/a wide variety of other chips and boards...

pmmarquez, 2023-07-10:
Hello @earlephilhower, first of all thanks for the awesome work you have made in this project , and for sharing with the community.

The CY8C9520A was tested so far using the I2C API of the pico-sdk and works perfectly, we discover the issue after adding a ATECC608A over the same I2C, this chip was intended to be used with the Arduino library and includes in a wakeup method using beginTransmission(adrr)-endTransmission() without any write (even this is the case mentioned at the beginning on #38), we were not able to make the ATECC608A work when sharing the bus with the CY8C9520A and later we notice the same was happening when running the Arduino i2c_scanner example, we identify that _probe was the source of the issue and after applying the fix everything works the i2c_scanner and the "0-len" wakeup of the ATECC608A.

This 1 byte dummy read is also the approach of the I2C bus_scan example of pico_examples.

The sniped shared before was just the quick fix for us not a proposal for changing, anyway later with more time we will keep investigating the issue, we are happy so far with the fix, because we definetly need to use the CY8C9520A.

This is maybe an unique or very rare case, but we though it was pertinent to share with the experience with the community.

earlephilhower, 2023-07-12:
No worries, we're just trying to figure out the right way to do things here.

But IMHO it seems like we already let you do a 1-byte read just fine, no?  Just do a SPI read in the code itself?  Would that not be a simple (and direct, portable) workaround in your app?

Converting a 0-byte *write* request into a 1-byte *read* request under the hood really feels like a technical debt to me...

pmmarquez, 2023-07-13:
@ [Controllino](https://www.controllino.com/) we are developing some RP2040 boards (probably we will make a PR to add support for our boards to arduino-pico in the future if posible), and sometimes choosing components depends on availability and costs so at this point we can not change the CY8C9520A port expander. 

This port expander(which cause the issue with _probe) is intendent to be handled under the hood, but the ATECC608A crypto is intended to be managed by the user using some library most likely [ArduinoECCX08](https://github.com/arduino-libraries/ArduinoECCX08/tree/master), so we have two options fix Wire or provide a fixed version of ArduinoECCX08, but we think on fixing Wire because we want also the classic Arduino i2c_scanner example to work.

Best Regards,

closed 2023-09-07 by earlephilhower