earlephilhower/arduino-pico #1604: Support for Tiny 2040

italic-r, 2023-07-25:
I'm finally getting around to learning about the 2040 chip and found this package for the 2040 family. The board I'm interested in is the Tiny 2040 by Pimoroni, mostly because it features USB-C and a reset button to reduce plug cycles. I tried using the generic 2040 but I couldn't get a simple LED fade project to work, no matter what pins I assigned. I found I could make the project work by specifying the Pico board and assigning the specific GPIO pin. The issue now is the Pico does not have as much flash as the Tiny (2MB vs 8MB). The Tiny does have a 2MB version, but I believe the standard is 8MB. This is the [product page](https://shop.pimoroni.com/products/tiny-2040?variant=39560012234835) for the 8MB.

earlephilhower, 2023-07-27:
You can use the Generic board and set an LED pin manually (since the default one may not correspond to the same one the Tiny chose).  You'll also need to make sure the flash chip matches (if in doubt, the SPI / 4 one is the most compatible...albeit slowest one) or the chip may not come out of reset and run your code properly.

italic-r, 2023-07-28:
> You can use the Generic board and set an LED pin manually (since the default one may not correspond to the same one the Tiny chose).

This is what I've been doing and I can make the Pico example work if I specify the pin number or `#include "boards/pimoroni_tiny2040.h"` to bring in some board definitions, but that's all (I'm not a C programmer).

> You'll also need to make sure the flash chip matches (if in doubt, the SPI / 4 one is the most compatible...albeit slowest one) or the chip may not come out of reset and run your code properly.

This must be where it's not behaving like I expect, because once I move to the Pico board in the IDE, programming the Tiny behaves like I expect. I don't quite know what you mean by matching flash chip. While in the generic board, I can choose any of the flash sizes (from 2MB to 16MB) and none of them run. I'm assuming there's much more to it than simply selecting that menu item.

I can build and run the examples from the `pico-examples` repo without issue. This requires manually invoking cmake with `-DPICO_BOARD=pimoroni_tiny2040` but everything behaves like I expect. Maybe I'm just over my head at the moment. I may just skip the IDE overall.

earlephilhower, 2023-07-28:
By flash chip, I meant the `Tools->Boot Stage 2` menu which is the different flash interfaces, not the flash size.  Each company, has their own special interface timing for high speed mode, and sometimes even the same company has different timings for different product lines.

What I'd recommend is going through each of the items in that menu, rebuilding and uploading until you get one that works properly.  (Or if you have good eyes, compare the names in the menu to the model # on the flash chip onboard!)

italic-r, 2023-07-28:
Oh! Well that's something I didn't notice. My chip says `Q64JVXG1Q` so I set it to `W25Q64JV QSPI /4` and it programs perfectly. Thanks!

PR #1672 (2023-08-25): Add Pimoroni Tiny2040
Fixes #1604.  Supports 2MB and 8MB revs
commit 21d2fb4a (2023-08-25): Add Pimoroni Tiny2040 (#1672)

Fixes #1604.  Supports 2MB and 8MB revs
closed 2023-08-25 by earlephilhower