earlephilhower/arduino-pico #1607: Breaking the peripheral interrupts when I2C ports swapped

earlephilhower, 2023-07-26:
This seems to break the peripheral interrupts(onReceive callback function never gets called), causing the RX buffer to fill, and SCL to be pulled low permanently causing the controller to freeze. TalkingToMyselfExample.ino does not work (even after swapping the Wire<>Wire1) assignments. Reverting back to the old pin_arduino.h fixes it. Not sure what the dependency is.

_Originally posted by @jfentonatnklabs in https://github.com/earlephilhower/arduino-pico/issues/1465#issuecomment-1650909920_

earlephilhower, 2023-07-26:
@jfentonatnklabs, I think you are right.  The proper IRQ *shim* handler is installed, but when it calls the actual object it doesn't look at the physical I2S port and assumes no swapping.
https://github.com/earlephilhower/arduino-pico/blob/67a07edcccb954c7ef8938ff64733f4a89eff893/libraries/Wire/src/Wire.cpp#L110-L116

jfentonatnklabs, 2023-07-26:
Thank you. I tried the fix in #1608 and it is working using the example file. One side note is the example file (TalkingToMyselfExample.ino) needs to have the pin numbers swapped between Wire<>Wire1 to account for the change in #1465  on the Adafruit Feather RP2040.

earlephilhower, 2023-07-26:
Thanks for the report!

commit bfe19b0d (2023-07-26): Handle swapped Wire IRQs properly

When we swap the Wire objects, we need the i2c0 IRQ shim to call
Wire1.onIRQ, not the usual Wire.onIRQ.  Same for i2c1 IRQ shim.

Fixes #1607
PR #1608 (2023-07-26): Handle swapped Wire IRQs properly
When we swap the Wire objects, we need the i2c0 IRQ shim to call Wire1.onIRQ, not the usual Wire.onIRQ.  Same for i2c1 IRQ shim.

Fixes #1607
commit 5fd4736d (2023-07-26): Handle swapped Wire IRQs properly (#1608)

When we swap the Wire objects, we need the i2c0 IRQ shim to call
Wire1.onIRQ, not the usual Wire.onIRQ.  Same for i2c1 IRQ shim.

Fixes #1607
closed 2023-07-26 by earlephilhower