earlephilhower/arduino-pico #1617: Could not find the package  ... framework-arduinopico  ... requirements for your system 'linux_x86_64'

MichaelMichaelMichaelMichaelMichael, 2023-07-30:
Hi, I'm trying to make esphome work under ubuntu by trying to run [Jeff Geerling's example](https://www.jeffgeerling.com/blog/2022/esphome-on-raspberry-pi-pico), but it fails with the error down-below. Do you have an idea what might be missing? 
Thank you in advance for your help! 
Michael

```
esphome on  rp2040 [?] via 🐍 v3.9.5 (esphome-build) 
❯ esphome run led-blink.yml --device /media/michael/RPI-RP2/
INFO Reading configuration led-blink.yml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
*************************************************************************************************************************************************************************
Obsolete PIO Core v6.1.4 is used (previous was 6.1.6)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
*************************************************************************************************************************************************************************
Processing rpi-pico (board: pico; framework: arduino; platform: platformio/raspberrypi @ 1.7.0)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing platformio/raspberrypi @ 1.7.0
INFO Installing platformio/raspberrypi @ 1.7.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: raspberrypi@1.7.0 has been installed!
INFO raspberrypi@1.7.0 has been installed!
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0
INFO Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-gccarmnoneeabi@1.90201.191206 has been installed!
INFO toolchain-gccarmnoneeabi@1.90201.191206 has been installed!
Tool Manager: Installing platformio/framework-arduino-mbed @ ~3.1.1
INFO Installing platformio/framework-arduino-mbed @ ~3.1.1
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduino-mbed@3.1.1 has been installed!
INFO framework-arduino-mbed@3.1.1 has been installed!
Tool Manager: Installing platformio/tool-rp2040tools @ ~1.0.2
INFO Installing platformio/tool-rp2040tools @ ~1.0.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-rp2040tools@1.0.2 has been installed!
INFO tool-rp2040tools@1.0.2 has been installed!
Tool Manager: Installing earlephilhower/framework-arduinopico @ ~1.20400.0
INFO Installing earlephilhower/framework-arduinopico @ ~1.20400.0
Error: Could not find the package with 'earlephilhower/framework-arduinopico @ ~1.20400.0' requirements for your system 'linux_x86_64'
```

maxgerhardt, 2023-07-30:
The PlatformIO package repository always deleted older versions of the `framework-arduinopico` when newer ones were uploaded. Now, they're even [completely removed](https://registry.platformio.org/search?q=framework-arduinopico) after a very lengthy [disagreement](https://github.com/platformio/platform-raspberrypi/pull/36) (read: "mud-wrestling) between PlatformIO and RaspberryPi Trading Ltd. Hence, the "could not find package" is true because it's really gone in the registry.

But, nothing to worry. The PIO registry is not needed at all for this to work. You can simply repoint the package `framework-arduinopico` to be taken from the git link `https://github.com/earlephilhower/arduino-pico/archive/refs/tags/2.4.0.zip` instead, giving you 2.4.0 (which is what `~1.20400.0` encodes). In some YAML file there should be some `platform_packages` declaration that controls this, I will have a look.

In any case though, even if it successfully uses 2.4.0, that's really outdated by now. Currently, it's 3.3.1. I think you're far better off in general if you take the current vanilla version of https://github.com/esphome/.

maxgerhardt, 2023-07-30:
No actually I think the blogpost is just wrong, sadly due to time advanced. When it tells you to use this fork of ESPHOME

>Use pip3's source install feature to install from the pull request fork: pip3 install git+https://github.com/jesserockz/esphome.git@rp2040

That's a branch now over 1000 commits behind mainlnie (https://github.com/jesserockz/esphome/tree/rp2040). May I ask you `pip uninstall esphome` again and use the `pip3 install git+https://github.com/esphome/esphome.git` instead?

earlephilhower, 2023-08-04:
No updates from the original filer and the referenced blog seems wildly out of date.  Closing.

closed 2023-08-04 by earlephilhower