earlephilhower/arduino-pico #1619: Serial Monitor not working in the Arduino IDE 2.1

Rastloser, 2023-07-30:
In a clean install the IDE Serial Monitor is not working.
To activate it, the following lines have to be added to the "platform.txt"-file:

# Required discoveries and monitors
# ---------------------------------
pluggable_discovery.required.0=builtin:serial-discovery
pluggable_discovery.required.1=builtin:mdns-discovery
pluggable_monitor.required.serial=builtin:serial-monitor

earlephilhower, 2023-08-01:
@Rastloser looks like you found and fixed a problem.  Would you like to make a PR?

FWIW, I think this is only a 2.1 issue.  I know for sure a clean install on 2.0 had no problem with the serial monitor because I did a lot of work w/it.

earlephilhower, 2023-08-04:
I've just downloaded 2.1.1 from the arduino.cc site under Ubuntu and run against `master`.  I still get the serial monitor (Ctrl-Shift-M and via Tools->Serial Monitor) listed, so I'm not sure what your original comment was referring to.  Are you saying it didn't work, or it wasn't available thru the menus, or something else?

Rastloser, 2023-08-04:
If I do that with a clean install on Windows 10, the following error-message appears as soon as I try to open the serial monitor:
`No monitor available for the port protocol serial. Could not connect to COM7 serial port.`

earlephilhower, 2023-08-04:
Sorry, I can't repro this on my Windows 10 machine, either.  I uninstalled any old IDEs then manually deleted all `arduino*` dirs in `C:\User\username\AppData\Local,LocalLow,Roaming` and `C:\Users\username\Documents`, then installed 2.1.1 from the Arduino website using the Windows Installer option.

After adding this core's URL and installing the tools, a simple sketch like
````
void setup() {
}
void loop() {
Serial.printf("hello\n");
delay(1000);
}
````
I just select the proper COM port (it sometimes changes on upload) and then Tools->Serial Monitor and it just works...
[image]

If you would like to make a PR for this, I guess I'd be OK to merge it, but I have no way of testing it as it seems unneeded no matter what version I try.

Since the AVR cores also need a Serial Monitor, and they are unconditionally installed on the first run of the IDE, I'm really not sure how it could not exist (esp. if it's listed in the menus!).

Rastloser, 2023-08-04:
I think that I found the problem:
for a clean install, I stripped all elements that were not exactly necessary (including the AVR cores), so that only the Pico boards were left. If I re-install the AVR cores again, the serial monitor starts working again even without the modifications described before, so the that the lines will probably affect only a very small minority of users...

commit 25052fed (2023-08-04): Update platform.txt to require auto-discovery and serial-monitor (#1631)

Fixes #1619

Requiring the auto-discovery tool and serial monitor if they have not already been included by other boards (namely if AVR cores have been uninstalled in the IDE).
closed 2023-08-04 by earlephilhower