earlephilhower/arduino-pico #1625: Change PID/VID Bug

meoinhan, 2023-08-02:
Dear  Sir,
I am trying to change the pid/vid at board.txt vid : 0x30fa pid : 0x1040 but when i check the pid is changed to 0x5040 i'm sure i entered it right. this is a good bug for some reason. Looking forward to some answer or help. Thanks a lot

maxgerhardt, 2023-08-02:
Are you sure you're changing the value of the macro?

https://github.com/earlephilhower/arduino-pico/blob/3cc5ac14ff5ab9fd2d0399c8d4a99c3451c5e818/boards.txt#L23-L42

meoinhan, 2023-08-02:
Yes i sure i changed

earlephilhower, 2023-08-04:
Changing Boards.TXT has no effect on the actual code that runs on the RP2040 which actually sets the VID/PID.

You need to set the `#define` macros at the build level as described by @maxgerhardt  and shown in
https://github.com/earlephilhower/arduino-pico/blob/1231317c9ca53c54db0ba306f6f8ac430f585a9c/cores/rp2040/RP2040USB.cpp#L49-L55

closed 2023-08-04 by earlephilhower

OrhanYigitDurmaz, 2023-08-07:
> Changing Boards.TXT has no effect on the actual code that runs on the RP2040 which actually sets the VID/PID.
>
> You need to set the `#define` macros at the build level as described by @maxgerhardt [...]

what about the name?

johndongus, 2023-08-23:
I am experiencing the same issue. Upon editing the values within the boards.TXT the values do change correctly, however if a PID starts with `0` Then it flash as `4`.

I have also tried editing the RP2040USB.cpp, after changing the defines I noticed no changes to the issue I've described. 

Note that VID is fine, even with a prefixed 0, and using a PID that starts with any value besides 0 seems to flash fine and result in the intended values.