earlephilhower/arduino-pico #1638: Mouse wheel smooth

LazaroFilm, 2023-08-08:
I'm using mouse.move(0,0,wheel) to simulate a mouse scroll, but the increment of the scroll is really big and erratic. My undertow that I need to modify the HID to change that behavior, but I'm not sure how to do that. 

[Code for my touchpad](https://github.com/LazaroFilm/touchscreen-trackpad)

maxgerhardt, 2023-08-09:
So you've tested `mouse.move(0,0,1);` in a minimal sketch and saw that even that mouse wheel value scrools too much and the increment should be finer?

LazaroFilm, 2023-08-09:
Exactly. the increment feels like one from a scroll wheel mouse where each notch is a big detent instead of a free wheel or trackpad "wheel" scroll where it's granular. Since I'm using a touchscreen to scroll I could have it as granular as the cursor move.

earlephilhower, 2023-08-09:
The HID report just contains +\- delta you pass in.  How far that delta moves something is up to the OS.  So, there’s nothing to be done here, sorry.

You could double check in the TinyUSB repo, but the HID descriptor seems very simple as well as the HID report format.

closed 2023-08-09 by earlephilhower

maxgerhardt, 2023-08-09:
Reminds me of https://www.youtube.com/watch?v=JzpUfRv45yY starting at about minute 8. Also https://pavelfatin.com/scrolling-with-pleasure/ with the "Mac OS" section.

Maybe you need a new USB HID descriptor that's more of a "trackpack" type thing? Or you could read the HID descriptor of your current, smooth scrolling mouse/ trackpad and compare the difference to the used HID descriptor, aka, https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/4a93c7d869852c03b198d74cefe87de4ba071eca/src/class/hid/hid_device.h#L220-L267.

LazaroFilm, 2023-08-09:
Thanks for the pointers. I'll dig further into this. And thank you for making arduino-pico!