earlephilhower/arduino-pico #1640: YD-RP2040 cannot access any more than 2MB

craigyjp, 2023-08-09:
I have recently been trying to flash some 4MB (25Q32 chip) and 16MB (25Q128 chip) YD-RP2040 boards with sample data for a project using Arduino IDE 2.1.1 and this package.

For some reason when selecting the boards and the correct layout 1MB/3MB and 1MB/15MB they fail with not enough space. If I change the device to an Arduino Feather 8Mb board split as 1MB/7MB it flashes OK. Obviously I would like to take advantage of the 16MB of flash or at least the 4MB boards.

c:/users/craig/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Craig\AppData\Local\Temp\arduino\sketches\D85D9B294E3A99E0405280EFC87CB732/Hagiwo_sample_drum_rp2040.ino.elf section `.rodata' will not fit in region `FLASH'
c:/users/craig/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 1847144 bytes
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

earlephilhower, 2023-08-09:
Sorry, you'll need to provide an MCVE because I'm not seeing any errors building samples on the 1/15MB variant.

````
... region `FLASH' overflowed by 1847144 bytes
````

seems to imply you've got a very large code image (i.e. a large constant array), since to overflow by 1.8MB on a 1MB space means ~2.8MB of flash program usage.

earlephilhower, 2023-08-09:
I didn't compile it, but just by inspection I can see a big problem:  The `sample.h` file has 48 instances of a `58,560` byte array.  That comes to `2,810,880` bytes just for arrays (which are code and not filesystem), so the error shown by the linker is correct.  You can't fit that in a 1MB code section.

You'll want to move those to separate files and read them in as needed (since you only have 256K RAM total).

earlephilhower, 2023-08-09:
And compiling with the Adafruit Feather RP2040 8MB (Sketch: 1MB   FS : 7MB) also fails properly (note the default is Sketch 8MB, FS 0MB on that board so if you just selected it w/o adjusting the ROM config it builds fine since it has 8MB of space available).

You can use any of the 4MB code configs to build on the VCC-GND board, just not ones with only 1 or 2MB of code space.

closed 2023-08-09 by earlephilhower

craigyjp, 2023-08-09:
I'm a little confused, this code originally was on an esp32-c3 and the author recommended a 1/3mb split to hold sample.h in the 3mb partition. Is it different on the rp2040?

earlephilhower, 2023-08-09:
Sorry, I haven't followed the ESP32 repo in a while.  All I can say is that any `const` global arrays are placed in the `.rodata` segment which is part of the actual code and in no way related to a filesystem.  Placing constants in the FS area would, in fact, be horrible and cause any FS accesses to overwrite your constants and generallly ruin your day. 😆 

Give the code 4MB to work in and you'll be fine here.

craigyjp, 2023-08-09:
Thanks for your help, I will experiment some more and see what happens.