earlephilhower/arduino-pico #1650: OpenOCD 0.12 from https://github.com/raspberrypi/openocd/tree/rp2040-v0.12.0

rgrr, 2023-08-20:
Hello Earle,

could you please update OpenOCD to version 0.12.  It is ready to compile at the usual raspberrypi repo.

earlephilhower, 2023-08-21:
As part of the process of doing a toolchain release, I need to get OpenOCD running in  GH CI so it can produce functional Win32/64 and MacOS binaries (cross-compilation doesn't work for them due to the `libUSB` dependencies).

Still debugging it, but it looks like GH's CI now has `aarch64` and not `x86_64` [Mac instances](https://github.com/earlephilhower/openocd/actions/runs/5920255994/job/16051297615) which is a) great and means I might get full ARM Mac binary builds for everything at some time, but b) breaks the GH CI auto-build I already have...

earlephilhower, 2023-08-21:
Nope, that was my fault.  No MacOS ARM CI instances yet, just me setting an artificial (incorrect) `--host` value, d'oh!  :sob:

Expect something this week if all else goes well!

rgrr, 2023-08-21:
You tagged this with "help wanted".  Can I be of any help?

earlephilhower, 2023-08-21:
Thanks for the offer, but I think I'm good now.

I initially thought I'd need someone w/a Mac to do some builds but as I found out the GH CI Mac instances work fine (but my script didn't!).  Win32/64 and Mac x86-64 builds ran fine, now I have some debugging on the cross-compilation for other Linux variants.  That's always the case since the build process and dependencies seem to change every release...

earlephilhower, 2023-08-21:
Please give #1657 a try.  I've got a clean build, finally, and updated to the latest GCC 10.5, GDB 10, and OpenOCD 0.12.  If CI passes I'll give it a few tests w/debugging and if all's well this will be in 3.4.0...

earlephilhower, 2023-08-21:
GMP 6.3.0 +GCC 10.5 seems to have a bug under Win32, boo. 😭

````
/workdir/repo/gcc-gnu/mpfr/src/init2.c:39: MPFR assertion failed: __gmp_bits_per_limb == (32 - 0)
In file included from c:\users\runneradmin\arduino_ide\hardware\pico\rp2040\system\arm-none-eabi\arm-none-eabi\include\c++\10.5.0\bits\specfun.h:46,
                 from c:\users\runneradmin\arduino_ide\hardware\pico\rp2040\system\arm-none-eabi\arm-none-eabi\include\c++\10.5.0\cmath:1935,
                 from c:\users\runneradmin\arduino_ide\hardware\pico\rp2040\system\arm-none-eabi\arm-none-eabi\include\c++\10.5.0\math.h:36,
                 from c:\users\runneradmin\arduino_ide\hardware\pico\rp2040\arduinocore-api\api\arduinoapi.h:47,
                 from C:\Users\runneradmin\arduino_ide\hardware\pico\rp2040\cores\rp2040/api/ArduinoAPI.h:2,
                 from C:\Users\runneradmin\arduino_ide\hardware\pico\rp2040\cores\rp2040/Arduino.h:28,
                 from D:\a\arduino-pico\arduino-pico\build.tmp\sketch\AnalogReadSerial.ino.cpp:1:
c:\users\runneradmin\arduino_ide\hardware\pico\rp2040\system\arm-none-eabi\arm-none-eabi\include\c++\10.5.0\limits:1669:7: internal compiler error: Aborted
 1669 |       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }
      |       ^~~
````

Feel free to try this PR if you're on Linux or Mac, but for Windows wait until I see if GMP 6.2.1 is any more stable.

rgrr, 2023-08-22:
> Please give #1657 a try. I've got a clean build, finally, and updated to the latest GCC 10.5, GDB 10, and OpenOCD 0.12. If CI passes I'll give it a few tests w/debugging and if all's well this will be in 3.4.0...

Hmmm... any instructions how to either obtain the build or build myself?

earlephilhower, 2023-08-22:
Hold off a day or so.  It turns out even the OpenOCD uploaded has some hiccups.

To try the PR once that's cleaned up at least, just clone this repo and check out the `gcc-10.5` branch and run `./tools/get.py` to get the new toolchain.

But, wait a bit for now until I get the 0.12 version working at least.  I've never had so obvious an internal GCC bug before so the 10.5 build is really kicking my butt...

rgrr, 2023-08-22:
> Hold off a day or so. It turns out even the OpenOCD uploaded has some hiccups.

Ok, I will stay tuned.

rgrr, 2023-08-23:
After updating and `get.py` I still get:

```
hardy@ntbox:/mnt/d/u/pico/arduino-pico$ ./system/openocd/bin/openocd --version
Open On-Chip Debugger 0.11.0-g8e3c38f78-dirty (2023-08-23-15:17)
...

hardy@ntbox:/mnt/d/u/pico/arduino-pico$ git status
On branch gcc-12.3
...
```

Any special reasons why you do not make the big step to 13.2?

Concerning your "head beating": I understand, that this does not create confidence into the build system.  Did you change anything in the build options?

earlephilhower, 2023-08-23:
Can you please triple check that you're running the right branch and re-ran `get.py` after changing the branch?  Here's what I see for Win64 using `strings` since I don't run Windows:

````
earle@amd:/tmp/233$ wget https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.0.0-1/x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip
--2023-08-23 10:49:12--  https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.0.0-1/x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip
Resolving github.com (github.com)... 192.30.255.112
Connecting to github.com (github.com)|192.30.255.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/335457897/c34357a3-c27d-493d-b884-5a093c5ed495?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230823%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230823T174913Z&X-Amz-Expires=300&X-Amz-Signature=eb36ada834c7c2165c120db5dcfdd5ea1e4b350cd10ea5abc20019d49970e877&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=335457897&response-content-disposition=attachment%3B%20filename%3Dx86_64-w64-mingw32.openocd-8e3c38f78.230823.zip&response-content-type=application%2Foctet-stream [following]
--2023-08-23 10:49:13--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/335457897/c34357a3-c27d-493d-b884-5a093c5ed495?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230823%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230823T174913Z&X-Amz-Expires=300&X-Amz-Signature=eb36ada834c7c2165c120db5dcfdd5ea1e4b350cd10ea5abc20019d49970e877&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=335457897&response-content-disposition=attachment%3B%20filename%3Dx86_64-w64-mingw32.openocd-8e3c38f78.230823.zip&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2412177 (2.3M) [application/octet-stream]
Saving to: ‘x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip’

x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip      100%[====================================================================================================================>]   2.30M  --.-KB/s    in 0.09s

2023-08-23 10:49:13 (25.0 MB/s) - ‘x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip’ saved [2412177/2412177]

earle@amd:/tmp/233$ unzip x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip
Archive:  x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip
   creating: openocd/
   creating: openocd/bin/
  inflating: openocd/bin/openocd.exe
  inflating: openocd/bin/libhidapi-0.dll
  inflating: openocd/bin/libusb-1.0.dll
   creating: openocd/share/
...
  inflating: openocd/share/openocd/scripts/chip/ti/lm3s/lm3s_regs.tcl
  inflating: openocd/share/openocd/scripts/chip/ti/lm3s/lm3s.tcl
   creating: openocd/share/openocd/scripts/test/
  inflating: openocd/share/openocd/scripts/test/selftest.cfg
  inflating: openocd/share/openocd/scripts/test/syntax1.cfg
  inflating: openocd/package.json
earle@amd:/tmp/233$ strings openocd/bin/openocd.exe | grep 0.12 | grep Open
Open On-Chip Debugger 0.12.0-g4d87f6dca (2023-08-22-03:02)
Open On-Chip Debugger 0.12.0-g4d87f6dca (2023-08-22-03:02)
````

````
earle@amd:/tmp/233$ sha256sum x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip
987ef40ef39ffc7e52c3dd5ab27fde44b015638f6a5a4e78a79c096d872e3afb  x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip
````

And the JSON file is linking to the riht version:
````
earle@amd:~/Arduino/hardware/pico/rp2040$ grep 8e3c38f78 ./package/*json
./package/package_pico_index.template.json:                     "url": "https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.0.0-1/aarch64-linux-gnu.openocd-8e3c38f78.230823.tar.gz",
./package/package_pico_index.template.json:                     "archiveFileName": "aarch64-linux-gnu.openocd-8e3c38f78.230823.tar.gz",
./package/package_pico_index.template.json:                     "url": "https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.0.0-1/arm-linux-gnueabihf.openocd-8e3c38f78.230823.tar.gz",
./package/package_pico_index.template.json:                     "archiveFileName": "arm-linux-gnueabihf.openocd-8e3c38f78.230823.tar.gz",
./package/package_pico_index.template.json:                     "url": "https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.0.0-1/i686-linux-gnu.openocd-8e3c38f78.230823.tar.gz",
./package/package_pico_index.template.json:                     "archiveFileName": "i686-linux-gnu.openocd-8e3c38f78.230823.tar.gz",
./package/package_pico_index.template.json:                     "url": "https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.0.0-1/i686-w64-mingw32.openocd-8e3c38f78.230823.zip",
./package/package_pico_index.template.json:                     "archiveFileName": "i686-w64-mingw32.openocd-8e3c38f78.230823.zip",
./package/package_pico_index.template.json:                     "url": "https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.0.0-1/x86_64-apple-darwin15.openocd-8e3c38f78.230823.tar.gz",
[... cut ...]
./package/package_pico_index.template.json:                     "url": "https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.0.0-1/x86_64-linux-gnu.openocd-8e3c38f78.230823.tar.gz",
./package/package_pico_index.template.json:                     "archiveFileName": "x86_64-linux-gnu.openocd-8e3c38f78.230823.tar.gz",
./package/package_pico_index.template.json:                     "url": "https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.0.0-1/x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip",
./package/package_pico_index.template.json:                     "archiveFileName": "x86_64-w64-mingw32.openocd-8e3c38f78.230823.zip",
````

rgrr, 2023-08-23:
```
hardy@ntbox:/mnt/d/u/pico/arduino-pico/tools$ ./get.py 
Platform: x86_64-pc-linux-gnu
Tool x86_64-linux-gnu.picotool-f6fe6b7.230823.tar.gz already downloaded
Extracting dist/x86_64-linux-gnu.picotool-f6fe6b7.230823.tar.gz
Tool x86_64-linux-gnu.openocd-8e3c38f78.230823.tar.gz already downloaded
Extracting dist/x86_64-linux-gnu.openocd-8e3c38f78.230823.tar.gz
Tool python3-via-env.tar.gz already downloaded
Extracting dist/python3-via-env.tar.gz
Tool x86_64-linux-gnu.arm-none-eabi-b3c2e32.230823.tar.gz already downloaded
Extracting dist/x86_64-linux-gnu.arm-none-eabi-b3c2e32.230823.tar.gz
Tool x86_64-linux-gnu.elf2uf2-6a7db34.230823.tar.gz already downloaded
Extracting dist/x86_64-linux-gnu.elf2uf2-6a7db34.230823.tar.gz
Tool x86_64-linux-gnu.pioasm-6a7db34.230823.tar.gz already downloaded
Extracting dist/x86_64-linux-gnu.pioasm-6a7db34.230823.tar.gz
Tool x86_64-linux-gnu.mklittlefs-4aca452.230823.tar.gz already downloaded
Extracting dist/x86_64-linux-gnu.mklittlefs-4aca452.230823.tar.gz
hardy@ntbox:/mnt/d/u/pico/arduino-pico/tools$ ../system/openocd/bin/openocd --version
Open On-Chip Debugger 0.11.0-g8e3c38f78-dirty (2023-08-23-15:17)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
hardy@ntbox:/mnt/d/u/pico/arduino-pico/tools$ git rev-parse --short HEAD
c41453c
```

Running on Debian.

earlephilhower, 2023-08-23:
Ha, you got me! 😆  `ntbox` with a `/mnt/d` path is excellent IT obfuscation. 😄   Anyway, will look into it.  The Win vs Linux builds have a different path and the checkout might have failed...

earlephilhower, 2023-08-23:
> Concerning your "head beating": I understand, that this does not create confidence into the build system. Did you change anything in the build options?

That's a bigger question than you might think.  For every GCC branch there are some things that need to be changed, either in the build process, or in the dependency versions, or via patches applied to the GNU or core code.  I've even had to update the Docker image because (it being the Internet Age and all) certain dependencies for cross-building are required to be grabbed online at runtime...and older revs are dropped by certain projects.  😭 

That said, I was able to install a clean Ubuntu 22.04 on my games box, `git checkout` the same pico-quick-toolchain that built the existing toolchain and see that #1437 compiles Windows to completion.  So the Docker image is OK, I just need to start bisecting the other bits of the Makefile, joy...

earlephilhower, 2023-08-25:
@rgrr, I just tested compiling AdvancedWebServer, bringing up OpenOCD and GDB, and the actual core still functions using #1670 .  You should be good to do with that branch `git checkout tool1`.

It seems RPI has discontinued support for the legacy PicoProbe (non-CMSIS) so I need to do some core tweaks to remove that option.  I just reflashed my existing debugger and things "just worked" when I used `./system/openocd/bin/openocd -f ./lib/picoprobe_cmsis_dap.tcl `

rgrr, 2023-08-25:
Ok, now OpenOCD is at 0.12!  I also used it successfully with my own CMSIS-DAP probe with an RP2040 and nRF52840 as targets.

It did not run immediately which was caused by some optimizations my probe is doing, namely DAP_PACKET_SIZE and COUNT.  That worked with 0.11 with 1024/2, with the 0.12 I'm currently at 256/1, which means some performance loss.  Not a big issue, but I will check that.

commit ade74986 (2023-08-25): GCC 12.3, GDB 12, OpenOCD 0.12, Picotool 1.1.2 (#1670)

Major toolchain update including:
* GCC 12.3
* GDB 12
* OpenOCD 0.12
* Picotool 1.1.2

* Fix MDNS infinite recursion

* Remove legacy Picoprobe

Fixes #1313
Fixes #1650
closed 2023-08-25 by earlephilhower