earlephilhower/arduino-pico #1652: Wire.setXXX while running to same pins when using multiple slave devices

coledeck, 2023-08-21:
Hi, 

I have a Pico W setup with 3 devices on I2C0 - LED driver + 2x PCF8574. I am not using the default pins, but I don't think it matters.

The library for the PCF8574 https://github.com/xreef/PCF8574_library/blob/master/PCF8574.cpp#L194 runs setXXX every time you call begin(). Which gets called twice, because I have 2 PCF8574s, so I have to do pcf1.begin() and pcf2.begin().

But as per I think #182 this raises a panic. 

Maybe consider not erroring in this case? It is setting the pins to the same pins it is already set to, so there's no risk of issues especially if the setXXX code just does nothing in this case.

Having 2 devices on the same I2C bus, even non-matching devices can have this issue depending on the library implementations.

```
 //                 SDA SCL
PCF8574 ioex1(0x20, 20, 21);
PCF8574 ioex2(0x21, 20, 21);

...

ioex1.begin(); // works fine
ioex2.begin(); // PANIC

```

maxgerhardt, 2023-08-21:
A `if (_sda == pin) return true;` check would save this case, but I'm not sure if other libraries don't run into different cases, same for SCL.

earlephilhower, 2023-08-21:
The way `Wire` is being used there isn't generally legal.  Calling `Wire.begin()` twice in a row could reinit things and generally make the object state inconsistent.  (Well, in this case it would ignore the 2nd `Wire.begin()` call because `Wire` was already running and hooked into the IRQs/etc.)

We can add in a `if (pinchange = noop) return;` I suppose which is safe any time and in this very specific case it may work (assuming that lib does *nothing* but `Wire.begin()` between your `ioex1.begin(); ioex2.begin()`)

coledeck, 2023-08-21:
I agree that it is not ideal to be calling these functions more than once, but it seems that most libraries for I2C devices call Wire.begin() at the very least, and some detect RP2040 and run setXXX. I did some testing and there didn't seem to be any issues with calling Wire.begin() multiple times.

commit 0c1b6ae6 (2023-08-21): Allow re-setting identical pins without panic()

Setting a pin to the current value is a no-op, not fatal.

Fixes #1652
PR #1655 (2023-08-21): Allow re-setting identical pins without panic()
Setting a pin to the current value is a no-op, not fatal.

Fixes #1652
commit 456b474a (2023-08-21): Allow re-setting identical pins without panic() (#1655)

Setting a pin to the current value is a no-op, not fatal.

Fixes #1652
closed 2023-08-21 by earlephilhower