earlephilhower/arduino-pico #1658: Multicore FIFO size

andreasdahl1987, 2023-08-22:
Hi!

Thanks for this amazing software! 

Is there any way to increase the size of the multicore FIFO buffer? I couldnt find any info on its size, but just pushing values into it, it seems like it stops at 8 32-bit values. I'd like it to be at least 100 times larger.

I'm running a serial communication with an application streaming LED data to one core, and a HID device on the other core. Doing both things on one core causes the application data steam to fail, so I'll have to keep them seperate. I'm also creating some internal LED data on the HID device core, so to put it all together and push to addressable LEDs, I need to transfer the LED data from one core to the other. I've tried just using a common array, but it seems both cores reading/writing to the array at the same time causes a lot of issues. Now trying with the multicore FIFO, the size of the buffer is making things difficult. 

Maybe pausing one core while writing to a common array is better than using multicore FIFO? How fast/responsive is this pause/resume?

Regards,
Andreas Dahl

maxgerhardt, 2023-08-22:
>it stops at 8 32-bit values

[image]

Exactly as [per datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf) page 309.

maxgerhardt, 2023-08-22:
Actually, scrap that. That's the PIO. That's not talking about the two Cortex-M0 cores.

But, don't they have the same view of RAM? Meaning you can create a regular global variable and some signaling variable (`volatile bool bufferReady`) and work on that buffer?

andreasdahl1987, 2023-08-22:
> Don't the two Cortex-M0 cores have the same view of RAM? Meaning you can create a regular global variable and some signaling variable (`volatile bool bufferReady`) and work on that buffer?

Ok I might have to revisit my initial attempt to make this work. So for instance a `uint32 ledArray [bufferSize]` to hold a 32-bit color for each LED, and the `volatile bool bufferReady` to go `true` when a full data package has been put in the buffer? 

Sending 8 values at the time is what I've been trying out without much success. A bigger buffer to take all the LED data and then arrange a transfer is probably a better idea yes, I'll give that a go as well.

Thanks for the quick reply :D

LinusHeu, 2023-08-22:
And afaik it's recommended to not use those hardware Inter-processor FIFOs (in user code, they're used by this core or the sdk).

earlephilhower, 2023-08-22:
Yes, the HW intercore FIFO is the only way for one core to interrupt the other.  In this core and the latest SDKs they're taken over by the infrastructure.  The `rp2040.fifo` implemented here is a simple SW-based one that's multicore safe but limited to the same size as the actual HW one.

Closing as this is not a core issue.

closed 2023-08-22 by earlephilhower