earlephilhower/arduino-pico #1660: Large inconsistencies with duration of file open and close (triggers watchdog)

bastian2001, 2023-08-22:
I am currently building a flight controller, and for that I am logging some data during flight. Each file has a header of ~150 Bytes, and then every ~300µs, another few bytes are written to the file. I use the following line to create and open the file: `blackboxFile = LittleFS.open(path, "w");`

There is a watchdog timer, fed every loop() cycle, which is once every few µs. With the WDT trigger set to 50ms, I can consistently trigger the watchdog on the creation of the 16th file, when I increase it to 100ms, the WDT triggers on the closing of the 28th file, and when I set the WDT to trigger at 250ms, I can consistently trigger it at the 43rd file. It is weird to me, that the time it takes the processor to create a file on the flash/LittleFS is inconsistent. When I set the WDT to 500ms, I haven't encountered an issue yet. However, for my application, a low WDT reset time is very beneficial.

I know for a fact, that it is actually due to LittleFS, because when I set the WDT to 500, and then create a big amount of files, say 150, then flash the MCU with a code with a low timeout, the program reboots on the first file already.

earlephilhower, 2023-08-22:
Interesting application, but I think you're abusing LittleFS in a way it's not really designed for.

This is a feature of LittleFS and not something we can affect here.  LittleFS' design goal is to be power fail safe across a wide variety of MCUs and flash devices, not to be high performance with many files.  There are no timing guarantees (especially since there are none on the flash device, either), only functionality.  In this case w/o WDT everything is running so it's technically running fine.

My guess is that this is related to it rewriting the entire root directory as you add files.  The same behavior is seen when you try and modify a file in the middle...it will dup the contents as you write leading to very low performance.  This is to guarantee a consistent state should power fail at any time.

One thing you might try is creating multiple directories and using them to store the new files, 10-20 per dir.  Or you could use a single file in append mode (which should be as safe as your 100s of tiny files thanks to the powerfail safety).

Other than that, you'll need to talk w/the upstream ARM LittleFS repo and see if they have any other suggestions.  You could also look at porting a non-flash-safe FS like FatFS (i.e. no wear leveling, no powerfail safety, but simple and fast) in your code.

bastian2001, 2023-08-22:
Thank you for your quick response and the two suggestions. How are directories implemented in LittleFS? To me it always seemed, like dirs are actually only slashes in filenames, together with some emulated directory functions. But they don't really seem to me like they could improve performance in this way. The other approach seems much more fit, as I don't mind if everything is in a big file. Each sequence will be separated anyway. Actually, in a realistic scenario, the flash won't even be enough for that many flights, so having that many files on the flash is unrealistic anyway. It's just the concern of unpredictability. I can not say for certain that it will always crash on the 16th file. Maybe a bigger first file will cause it to crash on the second file already and such things.

Either way, I will try both suggestions tomorrow, and report back :)

earlephilhower, 2023-08-22:
> How are directories implemented in LittleFS? To me it always seemed, like dirs are actually only slashes in filenames, together with some emulated directory functions

For ESP8266 SPIFFS that was the case, but in most FSes (and I believe in LittleFS) a directory is essentially just a file listing the names and indirection info/metadata of the files therein.

My thinking here is you're rewriting 10s of KB (and then erasing the old copy!) every time you add a new file to that long root list.  If the files are in subdirs, then you're only rewriting up to 10 dir entries and log10(n) root dir entries...

bastian2001, 2023-08-23:
Nice, that actually makes a ton of sense, and as such that is exactly what happens. Now, with a 100ms watchdog it only crashes at the 28th directory, rather than the 28th log file. Having implemented this approach, I now noticed another issue with LittleFS: the info function takes roughly 27ms (far too much during flight), when I have 100 files on the "drive". Sadly, I need to run this function quite often to tell me the available space. Because I don't completely know what's going on under the hood, I cannot just write to a file and expect the whole free space to be available to this file. 

I then implemented everything as a single file in append mode, with a 50KB safety margin and no continuous info() call. Rather than that, I now use file::size(), and compare that against the max free space before starting the logging (minus 50KB safety margin). That seems to work. file::size() seems to have a consistent execution speed of 13µs. Much better, though still not perfect. Maybe in the future I will track the file size manually, and increase the speed that way. Or maybe experimenting with lfs_file_write instead of File::printf will prevent the reboot once the flash is full without needing to continuously check the file size.

Too sad, that it is not possible to increase the emulated EEPROM size and do things manually. Another option for later is direct writing to the flash, while blocking the other core's operation. Wear leveling in my application isn't all that important, given that I will use the flash like a tape, and just write sequentially on most flash cells. Thus, all cells will age equally (except for sketch and settings space).

I'm gonna close this as not planned, because there's nothing you can do about it. Feel free to change it to completed, if you think that fits better or if GitHub mocks you for it :)

closed 2023-08-23 by bastian2001