earlephilhower/arduino-pico #1676: Missing include stdint.h in common.h

jpavieira, 2023-08-27:
In file https://github.com/earlephilhower/arduino-pico/blob/master/variants/generic/common.h 
I get many errors like this:
```
\framework-arduinopico\variants\generic\common.h:43:14: error: 'uint8_t' does not name a type
   43 | static const uint8_t D28 = (28u);
```

I could fix it by adding including stdint.h in that file.

earlephilhower, 2023-08-27:
That seems fine, would you like to make the PR for this?

I'm curious, though, how you're seeing this.  The variant's `.h` file is normally included by `Arduino.h` well after a `#include <stdint.h>` has happened.

jpavieira, 2023-08-27:
Sure, I will do it.
I'm also puzzled how this can happen. Anyway, the header has that dependency so it's not self-contained.

commit 13938115 (2023-08-27): Add stdint.h to generic variant common.h (#1677)

Fixes #1676
closed 2023-08-27 by earlephilhower