earlephilhower/arduino-pico #1682: USB - HID - loosing reports

benjaminaigner, 2023-08-31:
dear @earlephilhower ,

we experienced lost USB - HID reports when sending at a high rate or multiple devices at once (e.g. sending a keyboard & a mouse report at the same time).

I think it is related to 
https://github.com/earlephilhower/arduino-pico/blob/d0ac7f06b1fa75a4c098be3da543912e42a66d17/libraries/Mouse/src/Mouse.cpp#L48C34-L48C34

where the report is not sent when HID is is not ready.

Would it be feasible to replace the `if()` with a `while(!)`?
(maybe with an additional counter / us delay for a timeout)

Here is an example sketch, which triggers this error (warning: key "A" will be sticky)
[hid_test.txt](https://github.com/earlephilhower/arduino-pico/files/12484221/hid_test.txt)

earlephilhower, 2023-08-31:
Thanks for the clear description and failing sample.  Have you tried replacing the `if` with a `while` yourself?

My reasoning for the `if` was that if the HID device isn't up I didn't want to hang the machine , but maybe I misread the definition of `tusb_hid_ready`.  I think a timeout `while` including a USB connected check would make sense, as suggested.

earlephilhower, 2023-08-31:
@benjaminaigner I just tried your example with #1685 and it's looking good.  One `A` per second and one mouse right by 10px.  Please give it a try and report back.

benjaminaigner, 2023-08-31:
> My reasoning for the `if` was that if the HID device isn't up I didn't want to hang the machine , but maybe I misread the definition of `tusb_hid_ready`. I think a timeout `while` including a USB connected check would make sense, as suggested

Of course, I understand. In our case it would have happened when powering the Arduino with a USB powerbank and trying to send USB reports instead of BT reports.
Sorry, you were faster, I had no time to try it out with a `while()`

As always, congrats for your wonderful work, #1685 works as expected!

PR #1685 (2023-08-31): Ensure HID reports aren't dropped
Fixes #1682

Make the HID report wait up to 500ms for an existing one to go out before giving up sending a report.
commit b9c66ca0 (2023-08-31): Ensure HID reports aren't dropped (#1685)

Fixes #1682

Make the HID report wait up to 500ms for an existing one to go out
before giving up sending a report.
closed 2023-08-31 by earlephilhower