earlephilhower/arduino-pico #1691: Printing a zero value uint64_t number fails.

greiman, 2023-09-03:


earlephilhower, 2023-09-03:
Can you give an MCVE?  It works fine as far as I can tell:

````
void setup() {
}
uint64_t uz = 0;
void loop() {
    Serial.printf("%" PRIu64 "\n", uz);
    delay(500);
}
````
gives
````
0
0
0
...
````

greiman, 2023-09-03:
Sorry, I accident hit comment too soon.  see the above example.

earlephilhower, 2023-09-03:
Ah, thanks.  Arduino reinventing `vfprintf` and not actual `printf` which makes sense (the `Serial.printf` bit is handled by Newlib which I'm sure is good).

You might want to push this upstream to the main [ArduinoCore-API repo](https://github.com/arduino/ArduinoCore-API) since this is untouched Arduino.cc code.

earlephilhower, 2023-09-03:
https://github.com/arduino/ArduinoCore-API/issues/178

Already opened 9 months ago.

PR #1692 (2023-09-03): Fix Print::print(0ULL)
Move to a patched ArduinoCore-API revision.

Fixes #1691
commit 88cd4b52 (2023-09-03): Fix Print::print(0ULL) (#1692)

Move to a patched ArduinoCore-API revision.

Fixes #1691
closed 2023-09-03 by earlephilhower