earlephilhower/arduino-pico #1693: UART1 prints garbage.

the-maldridge, 2023-09-04:
I'm troubleshooting a problem with using Serial2/UART1 to provide an extra serial port in a design.  When using the following test program, I get garbage in my console that appears to be the right number of bytes.  At this point I'm inclined to think that there's something wrong in the underlying code that drives Serial2, so I'm looking for any help in figuring out what went wrong here.

```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial2.setTX(4);
  Serial2.setRX(5);
  Serial2.begin(300);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(250);
  digitalWrite(LED_BUILTIN, LOW);
  delay(250);
  Serial2.println("A");
}
```

maxgerhardt, 2023-09-04:
I can very easily check this with a logic analyzer on GP4 to see what's being output. Maybe the small baudrate is a problem because it can't find a right clock divider or delay number of for it? Did you test with another one like 9600?

earlephilhower, 2023-09-04:
I suggest you check your UART for proper configuration.  Default is 8N1, and with your example at 300bps I get a nice, decoded `A\r\n` using my logic analyzer.  Same at other baud rates:

[image]

closed 2023-09-04 by earlephilhower

maxgerhardt, 2023-09-04:
I can confirm the same results, compiling and uploading your sketch and feeding the GP4 signal to the RX of a USB-UART adapter (with the same common GND) gives the expected results at 300 baud.

[image]