earlephilhower/arduino-pico #1694: Possible WiFiClient hang when attempting to connect() to unreachable IP addresses

rw-r-r-0644, 2023-09-04:
It appears that WiFiClient's connect() function hangs indefinitely (rather than timing out) when attempting to connect to an unreachable IP address. The issue can be reproduced by modifying `host` in the simple WiFiClient example to an unreachable IP address (I've tested this with an IP in my same subnet).

I've traced this back to the `esp_delay` call in `ClientContext::connect`: 
https://github.com/earlephilhower/arduino-pico/blob/1231317c9ca53c54db0ba306f6f8ac430f585a9c/libraries/WiFi/src/include/ClientContext.h#L148
which for some reason doesn't appear to return after the set `timeout_ms` timeout when the IP is unreachable.
Curiously enough, while debugging the issue I've noticed everything works as expected when the `intvl_ms` interval is set to an higher value (e.g. `500` ms), so there could be a race condition somewhere (?).

**Details:**
_IDE_: Arduino 1.8.19
_Board_: Raspberry Pi Pico W
_Library version_: both 3.3.0 and 3.4.0 seem to have this issue (I did not test other versions)

earlephilhower, 2023-09-04:
Can you make a small MCVE showing the problem?  I'm not so sure `esp_delay` is the root here, it seems very simple on inspection https://github.com/earlephilhower/arduino-pico/blob/1231317c9ca53c54db0ba306f6f8ac430f585a9c/libraries/WiFi/src/include/ClientContext.h#L37-L44

rw-r-r-0644, 2023-09-04:
This should hang as described (`connect()` never times out as expected):
```C
#include <WiFi.h>

/* Set WiFi ssid/password */
#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

/* Example unreachable host on TEST-NET-1.
 * Unreachable hosts on normal, routable
 * IP blocks will also hang WiFiClient. */
const char* host = "192.0.2.42";
const uint16_t port = 17;

WiFiMulti multi;
WiFiClient wifiClient;

void setup() {
  Serial.begin(115200);
  while (!Serial);

  /* Connect to WiFi */
  Serial.print("connecting to WiFi ");
  WiFi.setTimeout(1000);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(500);
  }
  Serial.println(WiFi.localIP());

  /* Attempt to connect to unreachable TCP host */
  Serial.print("connecting to host ");
  Serial.println(host);

  if (!wifiClient.connect(host, port))
    Serial.println(".. timeout (this line never gets printed)");
  else
    Serial.println(".. success (host not unreachable)");

  Serial.println("disconnecting");
  wifiClient.stop();
}

void loop() {}
```

I agree, `esp_delay` is most likely not the root cause of the issue; the program hangs during the call to `esp_delay` in `WiFiClient::connect`, but if I had to guess it is probably one of the internal lwIP functions invoked by `sys_check_timeouts` in `esp_delay` that actually causes the bug.

Also for whatever reason while debugging I've noticed it appears to work when invoking SerialUSB functions during the `esp_delay` loop (changing the delay is not necessary, and when only changing the delay the issue remains), which seems fairly odd:
```diff
@@ -147,8 +147,9 @@ public:
         // will resume on timeout or when _connected or _notify_error fires
         // give scheduled functions a chance to run (e.g. Ethernet uses recurrent)
         esp_delay(_timeout_ms, [this]() {
+            Serial.write('.');
             return this->_connect_pending;
-        }, 1);
+        }, 500);
         _connect_pending = false;
         if (!_pcb) {
             DEBUGV(":cabrt\r\n");
```

earlephilhower, 2023-09-04:
Sorry, but I just tried your example and see no hang, no matter how many time I run it:
````
connecting to WiFi ...........192.168.1.131
connecting to host 192.0.2.42
.. timeout (this line never gets printed)
disconnecting
````
I also tried a routable, unused local IP and get the same response:
````
connecting to WiFi ..........192.168.1.131
connecting to host 192.168.1.194
.. timeout (this line never gets printed)
disconnecting
````

What core rev are you running?  Any special debug mode or options?

rw-r-r-0644, 2023-09-04:
Hmm, odd; perhaps it's network configuration related? I'm running core rev 3.4.0, with no debug mode options enabled
[image]

earlephilhower, 2023-09-04:
I'm not sure there's any network config differences.  We're both running `192.168.1.0/24` as far as I can see, and the connection to an unrouted/dead IP should be the same.

What would help, since I can't repro this, would be for you to hook up a Picoprobe and do a stack trace of the core when it's "stuck" in your setup using GDB.  That would at least give us somewhere in the code to look.  Right now, there's not really anything to be done here because there's nothing to go on yet.

rw-r-r-0644, 2023-09-06:
I've had a look at the MCVE through GDB. It appears that the Pico does not freeze, the sketch is still running; only SerialUSB stops working, which makes a lot more sense I guess. This probably explains why invoking `Serial.write()` inside of the loop avoided the problem, `SerialUSB::write()` internally invokes `tud_task()`.
Interestingly, when SerialUSB stops working, the device remains connected to the computer and on the host side there's no indication of the problem, it just doesn't deliver any more data.

earlephilhower, 2023-09-07:
That doesn't make much sense, unfortunately.  If the USB stack was hung (i.e. couldn't get the usb mutex so just no-op'd) then the USB connection would drop.  All USB code runs on the Pico proper so if it stops responding, you lose the serial ACM port and whatever else was running via Serial.

Can you dump the contents of the `Serial` object as well as the `__usb_mutex` one?  Might have a clue if some value looks amiss.

Another thing you can try is add `Serial.ignoreFlowControl()` to the beginning of your `setup` to short-circuit one possible branch of the `if` in `SerialUSB::write()`.

And you could try closing and re-opening the serial port monitor.  Due to power glitching or noise with a bad cable or USB port it might glitch and disconnect (I think you'll see something in `dmesg` about this) and reconnect afterwards as a new devnode.  Also try a new cable and USB port if possible, those are simple things to check.

I still can show no issues with your sample or other apps, and I know others are using USB and WiFi at the same time so I think there's something unique (lucky you!) in your setup...

earlephilhower, 2023-09-11:
@rw-r-r-0644 any update?

closed 2023-09-13 by earlephilhower