earlephilhower/arduino-pico #1695: Exception executing udisksctl. Exception: [Errno 2] No such file or directory: 'udisksctl'

dwrobel, 2023-09-06:
Installed versions:
- `arduino-pico` from `git` _(version: a385a4c1e5748c44743b86fa31c05309c0492df9)_
- `Arduino IDE` from flatpak:
```
$ flatpak info cc.arduino.IDE2 | grep Version
     Version: 2.2.1
```

An attempt to flash a `Temperature` sketch generates an error about missing `udisksctl` tool:

```
"/home/dw/Arduino/hardware/pico/rp2040/system/python3/python3" -I "/home/dw/Arduino/hardware/pico/rp2040/tools/uf2conv.py" --serial "/dev/ttyACM0" --family RP2040 --deploy "/home/dw/.var/app/cc.arduino.IDE2/cache/arduino/sketches/BA1CFB3C333AF819B31175B2A774E3BF/Temperature.ino.uf2"
Resetting /dev/ttyACM0
Converting to uf2, output size: 140800, start address: 0x2000
Scanning for RP2040 devices
Exception executing udisksctl. Exception: [Errno 2] No such file or directory: 'udisksctl'
Flashing /run/media/dw/RPI-RP2 (RPI-RP2)
Wrote 140800 bytes to /run/media/dw/RPI-RP2/NEW.UF2
Flashing /var/run/media/dw/RPI-RP2 (RPI-RP2)
Wrote 140800 bytes to /var/run/media/dw/RPI-RP2/NEW.UF2
```

I couldn't found `udisksctl` in the whole `pico` directory:
```
$ find ~/Arduino/hardware/pico -name udisksctl
$ 
```

earlephilhower, 2023-09-06:
`udisksctl` is part of `udisks` which is installed in most major distros.  It's the tool that handles things like mounting USB sticks...or RPi Pico IF2 bootloader volumes.

If running `udisksctl` from the command line doesn't work for you then you can just install that package.  If it does work but the message you show (which is non-fatal, obviously, since it still was able to copy the UF2 file) then there's a problem with the permissions on the flatpack.  Setting app permissions for Flatpacks and Snaps (especially) are something of a dark art, so as the README.md says we do recommend using the normal tarball installation for the IDE.

closed 2023-09-06 by earlephilhower

earlephilhower, 2023-09-06:
Sorry, but it's needed as the standard Linux way of getting USB sticks mounted when your system doesn't automount them.  Things here are working fine in your case, since the upload is working, so it can be safely ignored in your case.

dwrobel, 2023-09-06:
>  since the upload is working, so it can be safely ignored in your case.

So, perhaps there is no need to invoke the `udisksctl` at all and complain that it's not available.

earlephilhower, 2023-09-06:
> So, perhaps there is no need to invoke the udisksctl at all and complain that it's not available.

There definitely is, but in this FP case the Linux tools just aren't there so you get this warning.  If your OS was not set to automount USB sticks then uploads would fail, of course, since the FP couldn't actually run the tool needed to make that happen.

The logic is that we find the RPI Pico USB drive by `/dev/disk/by-ID/...` and then always attempt to mount it explicitly (to cover the case where the OS/GUI doesn't automatically do it).  That's a `udisksctl mount` operation.  In a normal install case where the standard Linux tools are available it's  is a safe and quiet noop if the drive is already mounted., or it will mount the drive as needed.

To do otherwise would require checking if the /disk/by-id/ was already mounted.  And the proper way to do that is to use...`udisksctl info`.  So it would still not work in a restricted flatpack.

You're lucky w/the Linux Flatpack, though.  The Windows store (equivalent  to FP or snap) Arduino install actually disables access to the UF2 drive and even the USB serial ports by default IIRC.  Sure, it's nice and secure, but it's also not very useful in that state. :laughing:

Sytoplis, 2025-04-13:
For anyone stumbling upon this, who is not able to get their flatpak to find udisksctl (like me, i really tried. Flatpak permissions are still black magic). I have a solution to get your system to upload to your Microcontroller WITHOUT needing your flatpak to access udisksctl. You still need udisksctl on your host system tho. So here is what I did:

First i gave the flatpak permissions to access my devices, so the paths `/media`, `/run/media` and `/mnt` should be added as permissions. You can do this by running
`flatpak override --user --filesystem=/media --filesystem=/run/media --filesystem=/mnt cc.arduino.IDE2`

When you get the the point where you want to upload your program, you select your board and your port, but before uploading, you run the udisksctl command externally yourself.
The code runs the following: 
https://github.com/earlephilhower/arduino-pico/blob/49397a7f3dd2f8a1ea68d5a4794aee4eaca0e4cf/tools/uf2conv.py#L275

So you open a terminal and run
`udisksctl mount --block-device /dev/disk/by-id/usb-RPI_RP2`_[some symbols here]_`-part1`
What symbols these are exactly, you need to figure out on your device. You can go into the directory and for me there was only one device that matched that description. Its also important that you DONT run this with sudo, as the device should be mounted as your user. When the command ran successfully you should see something like
`Mounted /dev/sdX1 at /run/media/[your username]/RPI-RP2`

Then you can run the upload in the IDE and it should work. You still should get one Error Message, as the search for the mount happens after udisksctl gets called, but it should still upload correctly.

SuperNinja-4965, 2025-04-13:
> For anyone stumbling upon this, who is not able to get their flatpak to find udisksctl (like me, i really tried. Flatpak permissions are still black magic). I have a solution to get your system to upload to your Microcontroller WITHOUT needing your flatpak to access udisksctl. You still need udisksctl on your host system tho. So here is what I did:
> 
> First i gave the flatpak permissions to access my devices, so the paths `/media`, `/run/media` and `/mnt` should be added as permissions. You can do this by running `flatpak override --user --filesystem=/media --filesystem=/run/media --filesystem=/mnt cc.arduino.IDE2`
> 
> When you get the the point where you want to upload your program, you select your board and your port, but before uploading, you run the udisksctl command externally yourself. The code runs the following:
> 
> [arduino-pico/tools/uf2conv.py](https://github.com/earlephilhower/arduino-pico/blob/49397a7f3dd2f8a1ea68d5a4794aee4eaca0e4cf/tools/uf2conv.py#L275)
> 
> Line 275 in [49397a7](/earlephilhower/arduino-pico/commit/49397a7f3dd2f8a1ea68d5a4794aee4eaca0e4cf)
>  cmd = ["udisksctl", "mount", "--block-device", os.path.realpath(rpidisk[0])] 
> 
> So you open a terminal and run `udisksctl mount --block-device /dev/disk/by-id/usb-RPI_RP2`_[some symbols here]_`-part1` What symbols these are exactly, you need to figure out on your device. You can go into the directory and for me there was only one device that matched that description. Its also important that you DONT run this with sudo, as the device should be mounted as your user. When the command ran successfully you should see something like `Mounted /dev/sdX1 at /run/media/[your username]/RPI-RP2`
> 
> Then you can run the upload in the IDE and it should work. You still should get one Error Message, as the search for the mount happens after udisksctl gets called, but it should still upload correctly.

Just a note as the flatpak maintainer, flatpak expects that udisks2 requests are made through dbus rather than attempting to call the binary directly. This should also work fine on any systems that utilise dbus and udisks outside of flatpak too. 

The simplest way to implement this within the library would be to use a command line tool such as `gdbus` which is available within flatpak to make the dbus calls. This would solve the issue for the arduino-ide flatpak where the udisksctl command is not available (and cannot be added to the flatpak).