earlephilhower/arduino-pico #1696: Pimoroni Tiny 2040 LED_BUILTIN is defined wrongly

Andy2No, 2023-09-08:
The Pimoroni Tiny 2040 board definition has LED_BUILTIN defined as 25, in v3.4.1, but does not have an LED on that pin.

It has an RGB LED (not NeoPixel) with connections to R, G and B on pins 18-20.  The pinout I'm referring to seems to imply that pin 19 (Green) should be the one used for LED_BUILTIN, from the way it's drawn, but any one of the three would do.

I've verified that the usual Blink sketch doesn't affect the LED via LED_BUILTIN (GP25) but flashes the Green LED if I use pin 19 instead.

[image]

Perhaps Pimoroni could say which of the three LED elements they'd prefer to be used for LED_BUILTIN.

earlephilhower, 2023-09-08:
Please feel free to update the variant.  The `git blame` for it is all on me now, and I'm happy to have others to point the finger at. :laughing:

I think I only did the variant because of the flash type and size being non-default.  I probably didn't look too closely at the schematic as you've seen.

Andy2No, 2023-09-08:
I made a pull request, then spotted a problem, so hold off on that one.

generic /  common.h refers to PIN_SPI0_SS but the Tiny 2040 doesn't have one - it doesn't have any physical header pins with that option.

The genetic / arduino.h has this:

#define PIN_SPI0_SS    (17u)

so I'll put that back as a dummy pin, unless there's a better way.

earlephilhower, 2023-09-08:
FYI, it's possible to swap SPI and SPI1, see some of the Adafruit boards where only SPI1 pinouts are available.

Andy2No, 2023-09-08:
Hmmm... going by their pinout, we can have SPI0 in one of two places, but they don't mention SPI1.

It only has GPIO0-7, plus the 4 ADC pins on GPIO26-29  like the ItsyBitsy RP2040 and WaveShare RP2040-Zero.  Going by the pinout of the standard Pico, it looks like there's no way to involve SPI1, or any SS pin.  I've also read that it can't have an SS pin, but I can't remember where, off hand.

I've amended the pins_arduino.h to put the dummy SS pin definition in.  I'm not sure if that's the best way to do it.  The alternative would be to not use generic / common.h, and copy in all but one line, but maybe that would break something else..

It's https://github.com/earlephilhower/arduino-pico/pull/1699

commit 585c31ef (2023-09-08): Pin definitions for Pimoroni Tiny 2040 (#1699)

Fixes #1696

Additional pin definitions for Pimoroni Tiny 2040, including using the Green LED element of the RGB LED as the default LED.

There are only 12 external header pins for GPIO, including 4 ADC pins, so there are less options for assigning pins than on a generic Pico RP2040. In particular, there can only be one SPI, and it can't have an SS pin but I've defined GPIO17, as on the pico, because the definition is used in generic / common.h.

BOOTSEL on the Pimoroni Tiny 2040 is connected to GPIO23. I don't know if that has any consequences f [... cut]
closed 2023-09-08 by earlephilhower