earlephilhower/arduino-pico #1700: Does LEAmDNS get me anything that WiFi.setHostname("picow"); doesn't?

morio, 2023-09-08:
I've been trying to run your WebServer library examples on the Pico W's second core. When ever I run MDNS.update(), I get a fault on the second core. It runs fine on the first core. I noticed from other code examples WiFi.h has a method `WiFi.setHostname("picow");`. When I run it I can access the webpage via http://picow. Is there any benefit to using LEAmDNS when I can already set the hostname with the WiFi library?
A

earlephilhower, 2023-09-08:
They're two different things.

The `WiFi.setHostname` sets a value the Pico sends in its DHCP request.  Some routers will take that name and add it to the local DNS so it works like you're seeing.  Other routers will not automatically make a DNS entry for that name, so looking up `picow` will not return an answer from the PC/phone.

MDNS is a distributed DNS where every device is responsible for sending its own answer when someone multicasts a "what IP is name XXXX?" request.  Even if the router doesn't move DHCP hostnames into its DNS cache, with MDNS and `.local` lookup you should still see the device by name.  (There's also add'l stuff like service advertisement included, too.)

> When ever I run MDNS.update(), I get a fault on the second core. It runs fine on the first core. 

Yes, that's expected,  All WiFi-related things need to run on Core0.  The WiFi stack is not protected against multiprocessor issues.  Don't call `.update` or any MDNS or WiFi or WiFiClient calls on Core1.

closed 2023-09-08 by earlephilhower

morio, 2023-09-08:
It seems I can run the WiFi Client and the WebServer library on the second core. I don't plan on mixing execution between the cores for those libraries. Off the top of your head do you think I might run into issues like race conditions even if I keep to the second core?

earlephilhower, 2023-09-08:
Yeah, I think you're going to have some issues.  The low-level IRQ and DMA handling for the CYW43 chip is all done on core 0.  So there is a possibility you get a new packet over the air on Core0 and it tries to stuff it into LWIP.  If Core1 is also doing something internal in LWIP at the same time, bad things will happen.