earlephilhower/arduino-pico #1701: version 3.4.1: Raspberry Pico W hangs (after a while) when running HTTPClient on both cores

calint, 2023-09-09:
I provide the sketch that reproduces the bug when running:
`while true; do wget -qSO- http://X.X.X.X/status;done`

Sketch:
```
#if !(defined(ARDUINO_NANO_ESP32) || defined(RASPBERRYPI_PICO))
#error "supports board Arduino Nano ESP32 and Raspberry Pico W"
#endif

#include <WiFi.h>
#include <NTPClient.h>
#include <HTTPClient.h>
#include <ArduinoJson.h>

#if defined(ARDUINO_NANO_ESP32)
#include <Preferences.h>
#endif

#include "secrets.h"  // defines WiFi login info 'secret_wifi_network' and 'secret_wifi_password'

constexpr const char* url_time_server = "http://worldtimeapi.org/api/ip";
constexpr const char* url_astros = "http://api.open-notify.org/astros.json";
constexpr const char* url_jokes = "https://v2.jokeapi.dev/joke/Programming";

WiFiServer web_server(80);

#ifdef RASPBERRYPI_PICO
const char* lookup_wifi_status_to_cstr(const uint8_t status) {
#else
const char* lookup_wifi_status_to_cstr(const wl_status_t status) {
#endif
  switch (status) {
    case WL_CONNECTED: return "connected";
    case WL_NO_SHIELD: return "no shield";
    case WL_IDLE_STATUS: return "idle";
    case WL_NO_SSID_AVAIL: return "no wifi network available";
    case WL_SCAN_COMPLETED: return "scan completed";
    case WL_CONNECT_FAILED: return "connect failed";
    case WL_CONNECTION_LOST: return "connection lost";
    case WL_DISCONNECTED: return "disconnected";
    default: return "unknown";
  }
}

#ifdef ARDUINO_NANO_ESP32
// code to run on second core
TaskHandle_t task_second_core;
void func_second_core(void* vpParameter) {
  setup1();
  while (true) {
    loop1();
  }
}
#endif

void hang() {
  while (true) delay(10000);
}

// setup first core
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  Serial.begin(115200);
  while (!Serial && millis() < 10000)
    delay(100);  // wait max 10 seconds for serial over usb

  Serial.printf("\nconnecting to '%s' with '%s'\n", secret_wifi_network, secret_wifi_password);

  WiFi.mode(WIFI_STA);
[... 248 lines cut ...]
  }
  
  client.stop();
  return true;
}

// loop on first core
void loop() {
  print_output_to_stream(Serial);
  // delay(10000);
}

// loop on second core
void loop1() {
  while (handle_web_server())
    ;
  delay(100);  // slightly less busy wait
}
```

secrets.h
```
constexpr const char* secret_wifi_network = "network";
constexpr const char* secret_wifi_password = "password";
```

earlephilhower, 2023-09-09:
This is expected.  Unfortunately the documentation was incorrect (#1702).

Only **core 0** may run WiFi or related code (docs used to say "only one core").  There's an  async process which handles LWIP `sys_check_timeouts` on core 0 that could end up causing re-entrancy in upper layers of the libraries, causing random crashes as you see.  You might just swap `setup/loop` and `setup1/loop1` and be okay assuming anything WiFi or TCP related is all on the one core.

closed 2023-09-09 by earlephilhower

calint, 2023-09-09:
Thank you for the quick reply.

I, of course, did not read the docs :)  My bad.